orbbec/OrbbecSDK_v2 #10: Reading sync port connections

Suedocode, 2024-10-11:
In the k4a library, you are able to query the status of the sync port connections, and analytically assign the sync mode settings
[is_sync_in_connected()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/classk4a_1_1device_acaba6c5d8a28965882a4703e8806810a.html)
[is_sync_out_connected()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/classk4a_1_1device_aec0191ecf75549f650af7ac61d0afbef.html)

```
  if (mDev.is_sync_in_connected()) {
    mConfig.wired_sync_mode = K4A_WIRED_SYNC_MODE_SUBORDINATE;
  } else if (mDev.is_sync_out_connected()) {
    mConfig.wired_sync_mode = K4A_WIRED_SYNC_MODE_MASTER;
  } else {
    mConfig.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
  }
```

I understand that Orbbec has a different syncing interface to utilize ethernet, but it'd be nice to at least query if a sync cable is connected in the first place, and then the software could distribute a master and subordinate configurations for multi-camera syncing. Currently, the sync connections have to be known outside of the software in order to decide if the sync configurations should be used. If the sync configurations are distributed but no sync signal exists, the cameras won't stream because they will never receive the trigger signal.

In the mean time, I will test using OB_MULTI_DEVICE_SYNC_MODE_SECONDARY_SYNCED to allow subordinate designated cameras to continue streaming when there is no sync cable connected. I'd prefer to use the more rigorous OB_MULTI_DEVICE_SYNC_MODE_SECONDARY in applicable situations, but it's difficult without being able to query the sync connection state.

Environment:
Ubuntu 22.04.5 LTS (x86_64)
g++ 11.4.0
OrbbecSDK-dev https://github.com/orbbec/OrbbecSDK-dev/commit/bd2d4dd41d6214c9d8bfe3365efa865d5fe8df9f
Femto: Firmware v1.2.9

hzcyf, 2024-10-12:
Thank you for your valuable feedback and suggestions.

1. The Femto Mega currently does not support the OB_MULTI_DEVICE_SYNC_MODE_SECONDARY_SYNCED mode.
2. We will discuss the additional issues and suggestions you’ve raised within our team. We’ll keep you updated with any developments.

Suedocode, 2024-12-07:
Our convention is to assign the first serial number by alphanumeric ordering as the PRIMARY, and the rest SECONDARY. This allows us to reliably set up the hardware according to that convention.

closed 2024-12-07 by Suedocode