orbbec/OrbbecSDK_v2 #12: Enumerating network devices on link-local network

Suedocode, 2024-10-13:
My hardware configuration is a wifi module proving the PC with internet (wlxec), and a network switch to connect the Femto cameras to the PC (eno1). In the Ubuntu "Wired Settings", I modified the IPv4 and IPv6 methods both to Link-Local Only:

```
$ ifconfig
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:a6:4b:05:0d  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

#THIS IS THE NETWORK SWITCH WITH FEMTOS, Link-Local address mask
eno1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.155.197  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::ea01:ba62:c14e:5e4e  prefixlen 64  scopeid 0x20<link>
        ether 6c:2b:59:f2:ec:00  txqueuelen 1000  (Ethernet)
        RX packets 2847  bytes 182208 (182.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 351  bytes 58874 (58.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  memory 0x92b00000-92b20000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1640209  bytes 1357321976 (1.3 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1640209  bytes 1357321976 (1.3 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

#THIS IS THE WIFI FOR INTERNET
wlxec1a5927c4ef: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.180  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::36fc:5f20:23c:e751  prefixlen 64  scopeid 0x20<link>
        ether ec:1a:59:27:c4:ef  txqueuelen 1000  (Ethernet)
        RX packets 3128902  bytes 2084167543 (2.0 GB)
        RX errors 0  dropped 491159  overruns 0  frame 0
        TX packets 2034334  bytes 1174517019 (1.1 GB)
        TX errors 0  dropped 545 overruns 0  carrier 0  collisions 0
```

I was hoping the OrbbecSDK would then magically enumerate them using `ob::Context::enableNetDeviceEnumeration(true)`, but as expected life is not so simple. Even just scanning the link-local network produces no results, though I recognize that we did not scan the entire space because it takes too long.

```
$ nmap -sn 169.254.155.0/24 -e eno1
Starting Nmap 7.80 ( https://nmap.org ) at 2024-10-13 13:56 CDT
Nmap scan report for 169.254.155.197 #this is our PC, seen on en01
Host is up (0.00032s latency).
Nmap done: 256 IP addresses (1 host up) scanned in 29.21 seconds
```

1. What exactly is the NetDeviceEnumerator looking for? This will help inform how we design and debug our hardware configuration.
2. Are Femto's compatible with [Link-Local](https://en.wikipedia.org/wiki/Link-local_address) type network connections, where they assign their own IPs as 169.254.0.0/16? This would be a very simple way to achieve what we want, without bothering with the whole DHCP overhead. Enumerating the space takes a while though...
3. Have Femto's been tested on network switches using `isc-dhcp-server`? We tried having Ubuntu itself host the DHCP network using `isc-dhcp-server` with no success, though it's possible that there was user error.
4. Should we use a router instead of a network switch so that the router itself will assign IP addresses (DHCP)?

Background:
For our use case, we want to keep the Femto sensors outside of the client's network:
1. We don't want any security implications, client network complications, or data burdens on their network.
2. Clients may have multiple devices, and splicing which sensors are assigned to which PC would be a source of infinite headache.
3. Our system should be as enclosed as possible when installed at a client location.

Environment:
Ubuntu 22.04.5 LTS (x86_64)
g++ 11.4.0
OrbbecSDK-dev https://github.com/orbbec/OrbbecSDK-dev/commit/bd2d4dd41d6214c9d8bfe3365efa865d5fe8df9f
Femto: Firmware v1.2.9

hzcyf, 2024-10-14:
This could be caused by multiple NICs（network interface controller）, you could try disabling the WIFI NIC and try that.

You asked several other questions, the responses are as follows: 
  1. NetDeviceEnumerator discovers Orbbec devices on the network through GVCP protocol, and gets the IP, MAC, PID, SN and other information of the devices. 2.
  2. Femto Mega (Warm reminder, Femto is another kind of device, you are using Femto Mega, it can be shortened to Mega) supports local network connection, after connecting the device via USB, you can configure the IP of the device via OrbbecViewer. 
  3. Femto Mega supports DHCP and tested with router，but we have not tested using Ubuntu as a DHCP host.
  4. Both are possible, Femto Mega supports DHCP dynamic IP and also supports static IP, but you need to make sure the network link and IP configuration are correct.

hzcyf, 2024-10-14:
In addition, you can try to connect to the device by entering the device IP through OrbbecViewer, this can help to analyze whether the problem is caused by the GVCP network device discovery code or network link and network configuration errors.
[image]

Suedocode, 2024-10-14:
I went out and bought a router, and connected that to my switch to server as the DHCP server. I cannot get the Megas to be assigned IP addresses within the subnet. In fact, they seem to be glued to 192.160.1.10 no matter what I do on my end. I am not done ruling out my own user error, but I'm struggling to understand what the issue is. I've done a factory reset on my router, set it to a subnet of 10.5.5.2-255, and yet both Megas insist on the exact same 192.160.1.10 IP address somehow. I think that shouldn't even be possible...

Anyway, still working on it. Thanks for your quick answers!

Suedocode, 2024-10-14:
[image]

hzcyf, 2024-10-15:
It is possible that the Mega's DHCP mode is not enabled, and the Mega device is configured as a static ip by default; you can enable the DHCP function through the OrbbecViewer tool after linking the device via USB.
[image]

Suedocode, 2024-10-15:
YES that was the problem! Thank you!

closed 2024-10-15 by Suedocode

Suedocode, 2024-10-15:
Is there an API to set this? I don't see any api or example that uses an OBNetIpConfig.

hzcyf, 2024-10-16:
> Is there an API to set this? I don't see any api or example that uses an OBNetIpConfig.

Yes, you can set it like this: `ob::Device::setStructuredData(OB_STRUCT_DEVICE_IP_ADDR_CONFIG, &netConfig, sizeof(OBNetIpConfig));`

We will consider adding an example for this in the future, thanks for the heads up.