orbbec/OrbbecSDK_v2 #13: ob::Device::setStructuredData throws blank exception but appears to succeed

Suedocode, 2024-10-16:
I am writing a tool to enable the dhcp setting on devices with this simple code:
```
OBNetIpConfig config;
memset(&config, 0, sizeof(config));
config.dhcp = 1;
dev->setStructuredData(OB_STRUCT_DEVICE_IP_ADDR_CONFIG, (const uint8_t*)&config, sizeof(config));
```
I get this exception:
```
terminate called after throwing an instance of 'ob::Error'
  what():  Request failed, device response with error, errorCode: 1024, msg: 
```
I also read the dhcp setting and print it before attempting to enable it. After running this tool and terminating with the exception, **the dchp setting appears to have enabled as desired**, I'm just not sure what the exception is trying to tell me.

Environment:
Ubuntu 22.04.5 LTS (x86_64)
g++ 11.4.0
OrbbecSDK-dev https://github.com/orbbec/OrbbecSDK-dev/commit/bd2d4dd41d6214c9d8bfe3365efa865d5fe8df9f
Femto: Firmware v1.2.9

hzcyf, 2024-10-17:
You are planning to change the DHCP-enabled IP configuration directly in the network connection state, right?

The problem should be that the firmware accepts the command and executes it immediately, making the network connection based on the old IP configuration unavailable and causing the SDK to fail to receive the response from the firmware.

I'll contact Femto Mega's firmware team about this issue and see if I can optimize it.

For now, you can just ignore the exception.

Suedocode, 2024-10-17:
These error were produced over the USB interface. It's not a blocking issue though, and we can ignore it.

hzcyf, 2024-10-21:
The Femto Mega firmware team has indicated that the errorCode: 1024 typically results from the device not being connected to a DHCP-enabled router or host. Once the device receives the DHCP configuration command, it will attempt to acquire an IP address from the router. If this process is unsuccessful, the error code will be triggered.

I have verified this on my end, and it is normal practice to first connect the device to the router before configuring it.

hzcyf, 2024-10-23:
Close as it is not a issue.

closed 2024-10-23 by hzcyf