orbbec/OrbbecSDK_v2 #18: Bug: Unable to create Spatial Filter

jb455, 2024-11-06:
An exception is thrown when creating a spatial filter with the code:
`auto spatialFilter = std::make_shared<ob::SpatialAdvancedFilter>();`
The message given is:
`Invalid filter name, cannot find filter creator for filter name: SpatialAdvancedFilter`

Tested with precompiled release 2.0.18.0 and latest commit to main on this repo (https://github.com/orbbec/OrbbecSDK_v2/commit/d6d044aa3b531048f33b45b3076a3f88f4ae638c), using a Gemini 336 on FW 1.3.70.

Note that it is possible to enable the filter using the Orbbec Viewer app (V2.0.18-main) without error.

Just out of interest I tried replacing "SpatialAdvancedFilter" with "SpatialFilter" at https://github.com/orbbec/OrbbecSDK_v2/blob/d6d044aa3b531048f33b45b3076a3f88f4ae638c/src/device/gemini330/G330Device.cpp#L1097 (and the following line) and it appears to fix the issue but I don't know if that will cause problems elsewhere.

hzcyf, 2024-11-06:
Thanks for the feedback, can you confirm that you have copied the extensions libraries to your application working directory？Perhaps this is due to the SDK not successfully finding and loading the extensions libraries.

jb455, 2024-11-06:
Hi, thanks for the quick reply.
I do have the extensions libraries in my output directory - to confirm, we should maintain the /extensions/... structure, and not have all the libraries copied to the root output directory?

I added the `auto spatialfilter...` line to the `ob_depth` sample and met the same error as in my code, you could try that to replicate the bug yourself (I'm assuming the samples have all libraries added correctly).

jb455, 2024-11-06:
I just realised I get the same error when trying to initialise `ob::NoiseRemovalFilter`, `ob::DisparityTransform` and `ob::TemporalFilter` the same way too, though `ob::DecimationFilter` and `ob::HdrMerge` work.
Is the only intended way to get filters to go via `sensor->createRecommendedFilters()`? We want to enable filters and set parameters individually in our app, it would be possible to loop through the FilterList and pick out which filters we are interested in but feels like a bit of a hack.

hzcyf, 2024-11-06:
[image]
I used precompiled release  v2.0.18  to add the code you mentioned and tested it without problems.  

---
There is one piece of information that we haven't updated the documentation on:  for commercial reasons, some filters are tied to our devices and need to be created after a device or pipeline created. This may be the cause of your problem.

hzcyf, 2024-11-06:
> Hi, thanks for the quick reply. I do have the extensions libraries in my output directory - to confirm, we should maintain the /extensions/... structure, and not have all the libraries copied to the [...]

Just keep the `/extensions/...` structure.

jb455, 2024-11-06:
> There is one piece of information that we haven't updated the documentation on: for commercial reasons, some filters are tied to our devices and need to be created after a device or pipeline created. This may be the cause of your problem.

Ah, that fixed it, I just moved filter initialisation to after pipeline start. Thanks for your help!

hzcyf, 2024-11-06:
OK, glad to hear it's fixed.  This is still an oversight in our documentation and interface comments, and we'll add a description of it in the next release before we close this issue.

chenin-wang, 2025-05-22:
moved filter initialisation to after pipeline start. 
https://github.com/orbbec/OrbbecSDK_v2/blob/main/docs/tutorial/installation_and_development_guide.md

[image]