orbbec/OrbbecSDK_v2 #19: CoordinateTransformHelper produces bad curvy results compared to OrbbecViewer

Suedocode, 2024-11-20:
I am having trouble getting the same 3d cloud results from OrbbecViewer using the CoordinateTransformHelper.

To compute the 3d points, we use this:

`
OPPoint2f p2{depth_frame_x, depth_frame_y}; //iterate through all depth frame pixels to compute 3d cloud
float d = depth_data[i]; //convert depth value in frame from uint16_t to float

auto depth_intr = depth_stream_profile->getIntrinsic();
auto depth_ext = depth_stream_profile->getExtrinsicTo(depth_stream_profile);
OBPoint3f p3{};
CoordinateTransformHelper::transformation2dto3d(p2, d, depth_intr, depth_ext, &p3);

pcl::PointXYZRGB p;
p.x = p3.x*.001;
p.y = p3.y*.001;
p.z = p3.z*.001;
add_point_to_cloud(p); //skipping details here
`

It results in a curvy ground result.
[image]

Meanwhile, here's the OrbbecViewer results. The ground is flat and consistent.
[image]

I need guidance on how to reproduce the clean OrbbecViewer clouds using the CoordinateTransformHelper functions.

Environment:
Ubuntu 22.04.5 LTS (x86_64)
g++ 11.4.0
OrbbecSDK-dev https://github.com/orbbec/OrbbecSDK_v2/commit/bd2d4dd41d6214c9d8bfe3365efa865d5fe8df9f
Femto: Firmware v1.2.9

hzcyf, 2024-11-21:
For OrbbecViewer, we utilize the `ob::PointCloudFilter` filter to convert depth data into point clouds, which includes distortion correction. Therefore, it is also recommended that you use the `ob::PointCloudFilter` filter.

If your application requires a transformation similar to `CoordinateTransformHelper::transformation2dto3d` for depth to point cloud conversion, we currently do not offer a version with built-in distortion correction. This functionality exists within the SDK's internal code, but we have not yet completed the API wrapping and testing. We will provide this feature as soon as possible and notify you upon its release.

MPierer, 2024-11-21:
Hi, maybe if it's urgent you can just copy the code from _src/shared/utils/CoordinateUtil.cpp/hpp_ and use it directly.
I did this to speed up the back-projection 3D->2D (e.g. cropped/filtered point cloud back to depth image).
Calling it the CTH class for a 640x480 PCL was way to slow, trashing cache and re-fetching intrinsics/extrinsics on every call.
The same is true for the PointCloud filter (file: _src/filter/publicfilters/PointCloudProcess.cpp_) You could then pass your saved intrinsics.
Thanks Orbbec for open-sourcing it!

Suedocode, 2024-11-21:
Wait, why isn't `src/shared/utils/CoordinateUtil.hpp` simply moved to `include/libobsensor/hpp` so its interface gets exported as part of libOrbbecSDK.so?