orbbec/OrbbecSDK_v2 #22: Library files for development

dbansal-chwy, 2024-11-26:
In the development guide what all library files do we have to copy and what should be structure?

Sirudoi, 2024-11-29:
Thanks for your feedback! Based on your issue and the log information:

> [11/26 17:25:47.090569][warning][89933][ObException.hpp:48] Depth engine create and initialize failed,retCode204 terminate called after throwing an instance of 'libobsensor::io_exception' what():  Depth engine create and initialize failed,retCode204

It seems that you encountered an issue while initializing the DepthEngine when using the Orbbec Femto Bolt on a Linux system. A solution for similar issues can be found in https://github.com/orbbec/OrbbecSDK-K4A-Wrapper#qa, specifically the third point. You might find a solution there. If the issue persists, please provide more details or relevant source code so we can assist you further.

dbansal-chwy, 2024-11-29:
Hi @Sirudoi, thank you for the response, but I don't face this issue when running the examples of the SDK,  shouldn't it give the same error there as well?

dbansal-chwy, 2024-11-29:
And for the following link, what all libraries should I copy:
https://github.com/orbbec/OrbbecSDK_v2/blob/main/docs/tutorial/installation_and_development_guide.md in linux

Sirudoi, 2024-11-30:
> Hi @Sirudoi, thank you for the response, but I don't face this issue when running the examples of the SDK, shouldn't it give the same error there as well?

Before starting the depth stream in your project, have you called any OpenGL-related APIs? The issue might be related to this. The SDK samples does not involve any OpenGL-related operations. 

Regarding your second question, when using the SDK, you need to copy the `orbbecsdk/lib/extensions` folder to the same directory as your executable. Please make sure to maintain the directory structure of the `extensions` folder, as shown below:
> .
├── build
├── extensions
│   ├── depthengine
│   │   ├── libdepthengine.so
│   │   └── libdepthengine.so.2.0
│   ├── filters
│   │   ├── libFilterProcessor.so
│   │   └── libob_priv_filter.so
│   ├── firmwareupdater
│   │   └── libfirmwareupdater.so
│   └── frameprocessor
│       └── libob_frame_processor.so
└── YourExecutable

I also noticed that in your `CMakeLists.txt`, you are copying the extensions folder to `${CMAKE_BINARY_DIR}/extensions/depthengine`, which could cause some issues. It would be better to modify it to `${CMAKE_BINARY_DIR}/extensions`.

dbansal-chwy, 2024-12-02:
I user glxgears to test the OpenGL, works fine for me. Also, I was trying pyorbbecsdk as well on WSL2-Ubuntu22-04, but I face same issues there as well.

Sirudoi, 2024-12-02:
> I user glxgears to test the OpenGL, works fine for me. Also, I was trying pyorbbecsdk as well on WSL2-Ubuntu22-04, but I face same issues there as well.

To avoid the OpenGL context interfering with the initialization of the depth engine, you can try the following steps:
```C++
ob::pipeline pipe
std::shared_ptr<ob::Config> config = std::make_shared<ob::Config>();
config->enableVideoStream(OB_STREAM_DEPTH);

GLFWwindow *currentContext = glfwGetCurrentContext(); // store the current context
glfwMakeContextCurrent(NULL);  // make current context to NULL

pipe.start(config); //  will initialize the DepthEngine

glfwMakeContextCurrent(currentContext); // restore the current context
```

dbansal-chwy, 2024-12-02:
Hi @Sirudoi , I tried your solution, no depth engine create failed error but the code gets stuck at the following:
ob_capture: ./src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == GLFW_TRUE' failed.

Sirudoi, 2024-12-03:
Thank you for your patience in trying, I have not encountered this problem myself. It’s possible that this solution caused a timing issue with GLFW initialization. If possible, could you provide the relevant code snippets to help us assist you better? This was an oversight on our part, and we will add a description of this issue in our future documentation.

dbansal-chwy, 2024-12-03:
HI @Sirudoi,
I can't share much of code but will do my best:
System: WSL2 - Ubuntu22-04 [All updated]

### NVIDIA-SMI:
**Tue Dec  3 08:54:16 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.57.02              Driver Version: 566.03         CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA RTX 5000 Ada Gene...    On  |   00000000:01:00.0 Off |                  Off |
| N/A    0C    P3             25W /   87W |       0MiB /  16376MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+**

### glxinfo -B

**name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA RTX 5000 Ada Generation Laptop GPU) (0xffffffff)
    Version: 23.2.1
    Accelerated: yes
    Video memory: 48540MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 4.2
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA RTX 5000 Ada Generation Laptop GPU)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
OpenGL shading language version string: 4.20
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.2.1-1ubuntu3.1~22.04.2**

Code Snippets:

`    std::cout<<"Setting up pipeline"<<std::endl;
    ob::Context ctx;
    ctx.setLoggerToConsole(OB_LOG_SEVERITY_INFO);
    mDev = ctx.queryDeviceList()->getDevice(0);

    try
    {   
        m_pipeline = std::make_shared<ob::Pipeline>(mDev);
        m_config = std::make_shared<ob::Config>();
        try{
            auto depthProfiles = m_pipeline->getStreamProfileList(OB_SENSOR_DEPTH);
            try
            {
                auto depthProfile = depthProfiles->getVideoStreamProfile(640, 576, OB_FORMAT_Y16, 30);
                m_config->enableStream(depthProfile);
            }
            catch(ob::Error& e)
            {
                std::cout<<"Selected profile was not found switching to default"<<std::endl;
                auto depthProfile = depthProfiles->getVideoStreamProfile(OB_PROFILE_DEFAULT);
                m_config->enableVideoStream(OB_STREAM_DEPTH);
                std::cout<<"Enabled Depth Stream"<<std::endl;

            }

        }
        catch(ob::Error &e){
            std::cerr << "function:" << e.getName() << "\nargs:" << e.getArgs() << "\nmessage:" << e.getMessage() << "\ntype:" << e.getExceptionType() << std::endl;

        }
        
        auto profiles = m_pipeline->getStreamProfileList(OB_SENSOR_COLOR);
        auto colorProfile = profiles->getVideoStreamProfile(1280, 720, OB_FORMAT_RGB, 15);
        std::cout<<"Enabling Streams"<<std::endl;

        m_config->enableStream(colorProfile);

        std::cout<<"Set up pipeline and config; starting pipeline"<<std::endl;
        m_pipeline->start(m_config);
        std::cout<<"Started Pipeline"<<std::endl;

    }
    catch(ob::Error &e){
        std::cerr<< "Error:" <<e.getMessage() << std::endl;
    }
   `
   
  
When I use this link for especially from point 2, I face this error.
When I build from source, and my code inside the repo and re-build it works fine, not sure what is the problem:
https://github.com/orbbec/OrbbecSDK

Also, for the pre-compiled libraries, the libraries in the extenstion/depthengine, I always get the error: "File too short" I have to copy the files from the source build which solves the error. Is it because the pre-compiled is from linux, and I am inside a WSL2 system?

Thank You.

Sirudoi, 2024-12-10:
Hello, I apologize for the delayed response.

Regarding the issue you mentioned, where you always encounter the "file too short" error when using the pre-compiled libraries, this could be caused by broken symbolic links.
Please make sure that you haven't extracted the Linux pre-compiled libraries on Windows, as this can break the symbolic links for the depth engine. Alternatively, you can try manually fixing this by running the following commands:
```bash
rm libdepthengine.so
ln -s libdepthengine.so.2.0 libdepthengine.so
```

Regarding the depth engine initialization issue: The depth engine creates an OpenGL context during initialization. If you call OpenGL functions before initializing the depth engine, it could cause an error due to multiple OpenGL contexts being created. A common solution to this is to release the OpenGL context before starting the stream:
```C++
auto currentContext = glfwGetCurrentContext();
glfwMakeContextCurrent(NULL);

m_pipeline->start(m_config); // start stream

glfwMakeContextCurrent(currentContext);
```
Could you help confirm whether the issue you encountered:
> ./src/posix_thread.c:64: _glfwPlatformGetTls: Assertion `tls->posix.allocated == GLFW_TRUE' failed.

is related to OrbbecSDK? Please try commenting out all OrbbecSDK-related code and keep only the OpenGL context storing and restoring logic. I've tried a few scenarios and couldn't reproduce the issue. 

We plan to resolve the multiple OpenGL context issue internally in the SDK in future versions, and we hope to improve this kind of experience going forward. Thank you for your support and usage!

closed 2024-12-10 by dbansal-chwy