orbbec/OrbbecSDK_v2 #35: How does timestamp sync actually work?

dbruning, 2025-01-21:
There's a great writeup about how the multi-camera sync trigger signal works [here](https://www.orbbec.com/documentation-mega/synchronize-multiple-femto-mega-devices/). I would love to see a similar writeup about how timestamp sync works.

Here's what I can gather so far:

 - The 8-pin GPIO port has [2 pins dedicated to timestamp sync](https://www.orbbec.com/documentation-mega/synchronize-multiple-femto-mega-devices/): pin 4 (TIMER_SYNC_OUT) and pin 7 (TIMER_SYNC_IN)
 - Each device has a [device timestamp reset config](https://orbbec.github.io/docs/OrbbecSDKv2/structob__device__timestamp__reset__config.html#a3021ee3e51e2a50b5d8c256e3f718df9) which describes how it will behave when receiving a timestamp reset signal (or being asked to send a timestamp reset signal)
 - We can call timestampReset() on a device to "reset the timer for calculating the timestamp for output frames to 0". Presumably this causes the device to also trigger its TIMER_SYNC_OUT signal, which will appear as TIMER_SYNC_In on secondary cameras.
 - We can also call timerSyncWithHost() on a device which means that the "timer of the device will be synchronized with the host".

Questions:
- How does timerSyncWithHost() work? What accuracy can we expect from that? It is just sending the host's current timestamp to the device for the device to adopt, and if so, wouldn't there be potentially a several-millisecond discrepancy still, due to network latency?
- Does timerSyncWithHost() use the hardware sync mechanism in any way? Should we call both timerSyncWithHost() and timestampReset(), and if so, in which order?
- When we call timerSyncWithHost(), do the subsequent frames then arrive with a timestamp of 0 and increasing from there? Or is the timer that's being reset get used as an offset against the device's own system timer somehow?
 - What is "global timestamp", e.g. isGlobalTimestampSupported() and enableGlobalTimestamp() on the [device class](isGlobalTimestampSupported())

Thanks for helping me understand this. My overall aim is to be able to match up frames from multiple cameras and be sure that they occurred at the same instant in time (subject to a known small depth frame time offset configured by multi-camera-sync)

zhonghong322, 2025-01-24:
Question 1:

How does timerSyncWithHost() work? What accuracy can we expect from that? It is just sending the host's current timestamp to the device for the device to adopt, 
and if so, wouldn't there be potentially a several-millisecond discrepancy still, due to network latency?
 
Ask: Yes, you are right,For different devices, the transmission delay may vary. For example, the Gemini 330 series has a delay of 1~2ms.

Question 2:
Does timerSyncWithHost() use the hardware sync mechanism in any way? Should we call both timerSyncWithHost() and timestampReset(), and if so, in which order?

Ask:
No, use Software time synchronization, timerSyncWithHost() and timestampReset() can be used together or individually.  

1. timerSyncWithHost means that the "timer of the device will be synchronized with the host", This interface is suitable for multiple cameras connected to the same Host PC.   
There is another interface in Context.hpp (enableDeviceClockSync(uint64_t repeatIntervalMsec)), which can periodically synchronize the clocks of all connected cameras.

2. Compared to timerSyncWithHost, timestampReset has an advantage: if multiple cameras are connected to different Host PCs, timestampReset offers higher precision. 
This is because timerSyncWithHost can have discrepancies due to differences in system time between multiple Host PCs.

3. With a multi-device sync hub, timestampReset only needs to be set on the Primary device, and the Primary device will configure the Secondary devices. 
you can call timestampReset() on a device to "reset the timer for calculating the timestamp for output frames to 0". Presumably this causes the device to also trigger its TIMER_SYNC_OUT signal, which will appear as TIMER_SYNC_In on secondary cameras.

Question 3:
When we call timerSyncWithHost(), do the subsequent frames then arrive with a timestamp of 0 and increasing from there? Or is the timer that's being reset get used as an offset against the device's own system timer somehow?

Ask: 
It does not start incrementing from 0; instead, it sets the Host's time to the device.

Question 4:
What is "global timestamp", e.g. isGlobalTimestampSupported() and enableGlobalTimestamp() on the device class?

Ask:
The global timestamp is calculated by first periodically retrieving the device time and the corresponding system time. 
Using linear fitting, two coefficients, A and B, are determined. Then, for each frame, the frame timestamp is obtained, and the global timestamp is calculated using the coefficients A and B.
GlobalTimestamp mark the exact moment when a frame was captured in device.

We recommend using the global timestamp. The global timestamp corresponds to the moment when the device captures the frame, making it more accurate for synchronization with other sensors, such as when you also have additional sensors like LiDAR.

dbruning, 2025-01-25:
OK, thankyou so much for that information, that's really useful.

closed 2025-01-25 by dbruning