orbbec/OrbbecSDK_v2 #36: Segmentation Fault

mikigom, 2025-01-22:
**Describe the bug**  
Running `./OrbbecViewer` or `./ob_depth` from `OrbbecViewer_v2.1.1_202412200848_b40e003_linux_x86_64` or `OrbbecSDK_v2.1.1_202412201026_5f278c73_linux_x86_64` results in a segmentation fault. However, `./ob_color` and `./ob_imu` execute without issues.

**Reproduction Steps**  
I followed the exact guideline using precompiled SDK package from the [doc](https://github.com/orbbec/OrbbecSDK_v2/blob/main/docs/tutorial/installation_and_development_guide.md). Let me know if you have any suspicious point for reproduction.

**Expected behavior**  
Both `./OrbbecViewer` and `./ob_depth` should execute successfully, similar to `./ob_color` and `./ob_imu`.

**Log Output**  

[OrbbecSDK.log.txt](https://github.com/user-attachments/files/18500021/OrbbecSDK.log.txt)

[OBViewer_LOG20250122135517.log](https://github.com/user-attachments/files/18500022/OBViewer_LOG20250122135517.log)

**System Environment:**

OS: Linux Ubuntu 24.04
Architecture: x86_64
GCC Version: 13.3.0

**Tool/SDK Version:**

- SDK: Version 2.1.1
- OrbbecViewer：: Version 2.1.1

**Device Information:**

- Model: Femto Bolt (SN: CL8NB4300XN)
- Firmware Version: 1.0.9

**Additional context**  
Add any other context about the problem here.

mikigom, 2025-01-22:
+ I tried on-build examples in docker container but I got the same Segmentation Fault error.

mikigom, 2025-01-22:
I changed part of `RawPhaseStreamer::parseAndOutputFrame` of `src/device/femtobolt/rawphase/RawPhaseStreamer.cpp` as follows ([Commit Link](https://github.com/mikigom/OrbbecSDK_v2/blob/fe7ecb61a2f9d1f9eab7df45e3ce263e4734edc1/src/device/femtobolt/rawphase/RawPhaseStreamer.cpp#L263-L281)). 
```cpp
    rawFrameSize = captureSize * inputInfo.nStreams;
    if(rawFrameSize > frame->getDataSize()) {
        LOG_ERROR("rawFrameSize exceeds actual frame->getDataSize(), dropping frame");
        return;
    }

    // We assume there's space after rawFrameSize for copying mipiHeadSize bytes
    size_t requiredSize = rawFrameSize + mipiHeadSize;
    if(requiredSize > frame->getDataSize()) {
        LOG_ERROR("rawFrameSize + mipiHeadSize exceeds actual frame->getDataSize(), dropping frame");
        return;
    }

    // The offset from which we copy the final row
    size_t lastRowOffset = static_cast<size_t>(inputInfo.nRows) * 2;
    if(rawFrameSize < lastRowOffset) {
        LOG_ERROR("rawFrameSize < lastRowOffset, inconsistent inputInfo or truncated frame");
        return;
    }]
```

It gives me the following log:
```
[01/22 06:45:12.104527][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.107239][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.108646][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.110101][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.111549][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.113007][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.121101][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
[01/22 06:45:12.121766][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
[01/22 06:45:12.126158][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
[01/22 06:45:12.129034][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.129040][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.136482][error][5148][RawPhaseStreamer.cpp:279] rawFrameSize < lastRowOffset, inconsistent inputInfo or truncated frame
[01/22 06:45:12.142169][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.147945][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.158360][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
[01/22 06:45:12.159560][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.162486][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.168299][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.169763][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.175625][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.178487][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.187380][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
[01/22 06:45:12.188678][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.191585][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.193039][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.194492][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.203253][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.204680][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.207592][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.209053][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.213408][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.219250][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.220680][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.222122][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.223578][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[... cut ...]
[01/22 06:45:12.238190][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
[01/22 06:45:12.241067][error][5148][RawPhaseStreamer.cpp:279] rawFrameSize < lastRowOffset, inconsistent inputInfo or truncated frame
[01/22 06:45:12.243948][error][5148][RawPhaseStreamer.cpp:265] rawFrameSize exceeds actual frame->getDataSize(), dropping frame
[01/22 06:45:12.245033][error][5148][RawPhaseStreamer.cpp:330] Process frame failed! Error code:1
```

Thus, it seems that the code is not getting the size of data it expects.

cool-wuzh, 2025-01-23:
> **Describe the bug** Running `./OrbbecViewer` or `./ob_depth` from `OrbbecViewer_v2.1.1_202412200848_b40e003_linux_x86_64` or `OrbbecSDK_v2.1.1_202412201026_5f278c73_linux_x86_64` results in a segmentation fault. However, `./ob_color` and `./ob_imu` execute without issues.
> 
> **Reproduction Steps** I followed the exact guideline using precompiled SDK package from the [doc](https://github.com/orbbec/OrbbecSDK_v2/blob/main/docs/tutorial/installation_and_development_guide.md). Let me know if you have any suspicious point for reproduction.
> 
> **Expected behavior** Both `./OrbbecViewer` and `./ob_depth` should execute successfully, similar to `./ob_color` and `./ob_imu`.
> 
> **Log Output**
> 
> [OrbbecSDK.log.txt](https://github.com/user-attachments/files/18500021/OrbbecSDK.log.txt)
> 
> [OBViewer_LOG20250122135517.log](https://github.com/user-attachments/files/18500022/OBViewer_LOG20250122135517.log)
> 
> Most importantly, the following log is the `gdb run` and `backtrace` log with DEBUG build config option:
> 
> ```
> (gdb) run
> Starting program: /root/OrbbecSDK_v2/build/linux_x86_64/bin/ob_depth 
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> [New Thread 0x7fffe9a00640 (LWP 20173)]
> [New Thread 0x7fffe9000640 (LWP 20174)]
> [New Thread 0x7fffe3e00640 (LWP 20175)]
> [New Thread 0x7fffe3400640 (LWP 20176)]
> [New Thread 0x7fffe2a00640 (LWP 20177)]
> [New Thread 0x7fffe2000640 (LWP 20178)]
> [New Thread 0x7fffe1600640 (LWP 20179)]
> [Thread 0x7fffe2000640 (LWP 20178) exited]
> [Thread 0x7fffe1600640 (LWP 20179) exited]
> [Thread 0x7fffe2a00640 (LWP 20177) exited]
> [New Thread 0x7fffe1600640 (LWP 20180)]
> [New Thread 0x7fffe2000640 (LWP 20181)]
> [New Thread 0x7fffe2a00640 (LWP 20182)]
> [New Thread 0x7fffe0c00640 (LWP 20183)]
> [Thread 0x7fffe2a00640 (LWP 20182) exited]
> [Thread 0x7fffe2000640 (LWP 20181) exited]
> [Thread 0x7fffe0c00640 (LWP 20183) exited]
> [New Thread 0x7fffe0c00640 (LWP 20184)]
> [New Thread 0x7fffe2a00640 (LWP 20185)]
> 
> Thread 13 "ob_depth" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0x7fffe0c00640 (LWP 20184)]
> __memmove_avx512_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:801
> 801     ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
> (gdb) backtrace
> #0  __memmove_avx512_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:801
> #1  0x00007ffff7a1e369 in operator()(std::shared_ptr<libobsensor::Frame const>) const (__closure=0x555555ae6fc0, frame=warning: RTTI symbol not found for class 'std::_Sp_counted_deleter<libobsensor::VideoFrame*, libobsensor::FrameBufferManager<libobsensor::VideoFrame>::acquireFrame()::{lambda(libobsensor::VideoFrame*)#2}, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>'
> warning: RTTI symbol not found for class 'std::_Sp_counted_deleter<libobsensor::VideoFrame*, libobsensor::FrameBufferManager<libobsensor::VideoFrame>::acquireFrame()::{lambda(libobsensor::VideoFrame*)#2}, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>'
> 
> std::shared_ptr<const libobsensor::Frame> (use count 3, weak count 1) = {...})
>     at /root/OrbbecSDK_v2/src/device/femtobolt/rawphase/RawPhaseStreamer.cpp:498
> #2  0x00007ffff7a1fdee in std::__invoke_impl<void, libobsensor::RawPhaseStreamer::initNvramData()::<lambda(std::shared_ptr<const libobsensor::Frame>)>&, std::shared_ptr<libobsensor::Frame> >(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/11/bits/invoke.h:61
> #3  0x00007ffff7a1fa9b in std::__invoke_r<void, libobsensor::RawPhaseStreamer::initNvramData()::<lambda(std::shared_ptr<const libobsensor::Frame>)>&, std::shared_ptr<libobsensor::Frame> >(struct {...} &) (__fn=...) at /usr/include/c++/11/bits/invoke.h:154
> #4  0x00007ffff7a1f632 in std::_Function_handler<void(std::shared_ptr<libobsensor::Frame>), libobsensor::RawPhaseStreamer::initNvramData()::<lambda(std::shared_ptr<const libobsensor::Frame>)> >::_M_invoke(const std::_Any_data &, std::shared_ptr<libobsensor::Frame> &&)
>     (__functor=..., __args#0=...) at /usr/include/c++/11/bits/std_function.h:290
> #5  0x00007ffff7635f59 in std::function<void (std::shared_ptr<libobsensor::Frame>)>::operator()(std::shared_ptr<libobsensor::Frame>) const
>     (this=0x555555ae6fc0, __args#0=std::shared_ptr<libobsensor::Frame> (empty) = {...}) at /usr/include/c++/11/bits/std_function.h:590
> #6  0x00007ffff76ebb99 in libobsensor::ObLibuvcDevicePort::onFrameCallback(uvc_frame*, void*) (frame=0x555555aeb0d8, userPtr=0x555555ae6fb0)
>     at /root/OrbbecSDK_v2/src/platform/usb/uvc/ObLibuvcDevicePort.cpp:441
> #7  0x00007ffff7be1d7e in _uvc_user_caller (arg=0x555555aea990) at /root/OrbbecSDK_v2/3rdparty/libuvc/src/src/stream.c:1273
> #8  0x00007ffff66abac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
> #9  0x00007ffff673d850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
> ```
> 
> **System Environment:**
> 
> OS: Linux Ubuntu 24.04 Architecture: x86_64 GCC Version: 13.3.0
> 
> **Tool/SDK Version:**
> 
> * SDK: Version 2.1.1
> * OrbbecViewer：: Version 2.1.1
> 
> **Device Information:**
> 
> * Model: Femto Bolt (SN: CL8NB4300XN)
> * Firmware Version: 1.0.9
> 
> **Additional context** Add any other context about the problem here.

Hello, Mikigom.

According to the log from the Orbbec SDK, your FemtoBolt device is currently in a USB 2.1 connection state. Please check the product manual for FemtoBolt, which states that the device does not support USB 2 connections. I recommend trying to reconnect it as USB 3.x and then testing again.

mikigom, 2025-01-23:
@cool-wuzh Thank you for your support. Your suggestion was really helpful for my problem. Just after connecting with USB 3.0, the latency is completely gone.

closed 2025-01-23 by mikigom