orbbec/OrbbecSDK_v2 #38: Can't configure Femto Bolt Depth camera synchronisation from Orbbec Viewer

DavidRobb, 2025-02-04:
**Describe the bug**  
I have two Femot Bolt depth cameras updated to firmware 1.1.2 and connected via multisync cables and a daisy chain hub. I am trying to configure the depth trigger delay but the secondary device depth delay is being ignored whe using Orbbec Viewer version 2.1.1-main

**Reproduction Steps**  
Connect the two cameras via USB and multi-sync cable config.

Run two instances of the Orbbec viewer. Connect one to the primary device and the other to the secondary.

Run Orbbec Viewer and connect to secondary device.

Enter Device Control . Synchronization Configuration:
    Set mode to secondary
    Depth Delay to 160uS
    Write settings

Viewer says multi-sync config succeed! but the Depth delay value is reset ot 0uS

On starting depth stream, output is correctly paused until the Primary stream is started. However, on starting the primary stream, depth is started on the secondary but suffers from interference as there is no delay between Primary and Secondary.

**Expected behavior**  
I expect the Depth delay setting to be stored and used to separate the image captures.

**System Environment:**

Windows 11
 x86_64

OrbbecViewer_v2.1.1_202412201710_b40e003_win_x64

**Device Information:**

- Model: Femto Bolt
- Firmware Version: 1.1.2

**Additional context**  
Setting the delay using the k4aviewer appears to work OK. My final objective is to set it using the SDK but this also seems to suffer the same problem as the Orbbec Viewer.

zhonghong322, 2025-02-05:
For Femto Mega and Femto Bolt devices, you can configure color delay. For multi-camera synchronization configuration, please refer to this document.

https://www.orbbec.com/docs-general/set-up-cameras-for-external-synchronization_v1-2/

DavidRobb, 2025-02-05:
I'm following these instructions, specifically Appendix IV for the Femto Bolt but

On my primary

[image]

The trigger out enable control is off and disabled. (But don't think this is an issue)

On my secondary, on opening it comes up as:

[image]

Which says Mode secondary_synced even though this is not one of the drop down options

When I change this to select Secondary and put in a value of 160us for depth delay then press write:

[image]

It says write mutli sync succeeded but changes the depth delay to 0us.

On starting the depth stream it does wait for the primary stream to also be started but the picture looks like the depth cameras are interfering (increased and pulsing noise present)

It does appear to work properly if I follow these instructions:
[k4a wrapper multisync](https://www.orbbec.com/documentation/synchronize-multiple-femto-bolt-devices/)

so the k4a wrapper somehow makes it work properly.

But my objective is to get it to work from the SDK and this shows the same problem as the Orbbec Viewer. Please advise how I can make it work.

DavidRobb, 2025-02-17:
Any comments re the above?

zhonghong322, 2025-03-08:
I found an issue in the product documentation and will sync with the product team. Please try testing with the color delay.
 [image]

kopeto, 2025-07-16:
I found that you need to set the trigger2ImageDelayUs in order to modify the depthDelayUs. I don't know this is working in all orbbec devices, it worked for me in my femto bolts.

I think the documentation should be updated or it should point out this issue( I found it through trial and error )