orbbec/OrbbecSDK_v2 #39: Astra2 pipeline stop crashes

GentBinaku, 2025-02-10:
**Describe the bug**  
Astra2 pipeline stop sometimes crashes

**Reproduction Steps**  
Pipeline start and stop in a while true loop crashes

**Expected behavior**  
Should start and stop normally

**Log Output**  
[OrbbecSDK.log.txt](https://github.com/user-attachments/files/18731807/OrbbecSDK.log.txt)

**System Environment:**

- OS: Windows 11
- Architecture: x86_64 
- Visual Studio/GCC：Version 19.42

**Tool/SDK Version:**

- SDK: 2.1.1
- OrbbecViewer：: 2.1.1

**Device Information:**

- Model: Astra2
- Firmware Version: 2.8.22

**Additional context**  
Sometimes the stop function takes so much time to stop.

zhonghong322, 2025-03-08:
I will test it according to your testing method.

zhonghong322, 2025-03-10:
I used version [v2.2.8](https://github.com/orbbec/OrbbecSDK_v2/releases/tag/v2.2.8 ) and performed stress testing according to your method. The test was normal. You can switch to this version and test it. @GentBinaku 

Stress tested for a weekend.
[image]

Abrahamh08, 2025-03-11:
I'm having a similar issue, where I pretty much have to restart my computer every couple of runs, and the very first time I used the Astra 2 in the viewer when I got it, I had to reboot my computer.

drhalftone, 2025-05-18:
I'm having an issue with the Astra 2 as well as the Femto Mega I.  I have more thorough testing on the Mega I where my code crashes consistently on the 9210th frame.