orbbec/OrbbecSDK_v2 #43: Access violation in object destruction after stopping pipeline.

MaksimDanilov, 2025-03-10:
**Describe the bug**  
After stopping pipeline sometimes I receive object access error. I simplified your examples (ob_point_cloud.exe) to catch that error. Just put to main exception handling and catch only one frames and then exit. Make object destruction before return statement. I guess pipeline received data after being stopped.

**Reproduction Steps**  
1. Run modified example (ob_point_cloud.exe) with debugger.
2. Catch depth frame.
3. Save to ply.
4. Destroy pipeline.

**Expected behavior**  
No frames received after pipeline destroyed.

**Log Output**  
Include relevant log entries or error messages that are related to the problem.You can attach log files if they are too large.

**System Environment:**

- OS: Windows 11
- Architecture: x86_64
- Visual Studio/GCC：Visual Studio 2022

**Tool/SDK Version:**

- SDK: 2.2.8

**Device Information:**

- Model: Microsoft Surface Pro 11.
- Firmware Version: Firmware revision number.

**Additional context**  

[image]

zhonghong322, 2025-03-12:
Can you provide the modified test code  you used?

MaksimDanilov, 2025-03-19:
> Can you provide the modified test code you used?

Sorry. Somehow missed message. Sure. I'll past file tomorrow.

MaksimDanilov, 2025-03-24:
[point_cloud.txt](https://github.com/user-attachments/files/19415538/point_cloud.txt)
@zhonghong322 Try this :-)