orbbec/OrbbecSDK_v2 #46: Device firmware update example segfault

jlucier, 2025-03-13:
**Describe the bug**  
Running the device firmware update example results in a segmentation fault.

**Reproduction Steps**  
1. Run the example
2. Select device
3. Select file path
4. Observe segmentation fault

**Expected behavior**  
The firmware to be updated successfully

**Log Output**  
```
$ ./bin/ob_device_firmware_update
Devices found:
--------------------------------------------------------------------------------
[0] Device: Orbbec Gemini 336L | SN: CP82841000BE | Firmware version: 1.2.81
---------------------------------------------------------------------------------
Please select a device to update the firmware, enter 'l' to list devices, or enter 'q' to quit:
Device index: 0

Please input the path of the firmware file (.bin) to be updated:
(Enter 'Q' or 'q' to quit):
Path: /home/jordan/Downloads/Gemini330_Release_1.4.60.bin
Firmware file confirmed: /home/jordan/Downloads/Gemini330_Release_1.4.60.bin

Upgrading device firmware, please wait...

Progress: 0%
Status  : File transfer in progress
Message : Ready to update firmware...
[1]    152096 segmentation fault (core dumped)  ./bin/ob_device_firmware_update

**System Environment:**

- OS: Fedora 40
- Architecture: x86_64
- Visual Studio/GCC：14.2.1

**Tool/SDK Version:**

- SDK: 2.1.1

**Device Information:**

- Model: Gemini 336L + 335L
- Firmware Version: 1.2.81

zhonghong322, 2025-03-28:
Can you test with the SDK version [v2.28](https://github.com/orbbec/OrbbecSDK_v2/releases/tag/v2.2.8)?

zhonghong322, 2025-10-25:
Has the issue been resolved using the new version of the SDK?