orbbec/OrbbecSDK_v2 #48: 2 cameras on one USB2 hub stalls depth channel

Chris-Pane, 2025-03-24:
**Describe the bug**  
When connecting two Gemini 330 cameras to one USB2 hub, no depth frame is received, color frames are partially corrupt. LibUVC receives more data than expected.

**Reproduction Steps**  
1. Connect two Gemini 330 cameras to one USB2 hub
2. Enable color stream 640x480 @ 30fps MJPEG
3. Enable depth stream 640x480 @ 30fps Y16

**Expected behavior**  
Depth and color frames are received with 30fps.

**Log Output**  
```
[03/24 08:44:53.603769][info][11517][FilterFactory.cpp:32] Registered 14 filter creators
[03/24 08:44:53.603776][info][11517][FilterFactory.cpp:34]  - Align
[03/24 08:44:53.603778][info][11517][FilterFactory.cpp:34]  - DecimationFilter
[03/24 08:44:53.603778][info][11517][FilterFactory.cpp:34]  - FormatConverter
[03/24 08:44:53.603779][info][11517][FilterFactory.cpp:34]  - FrameFlip
[03/24 08:44:53.603780][info][11517][FilterFactory.cpp:34]  - FrameMirror
[03/24 08:44:53.603781][info][11517][FilterFactory.cpp:34]  - FrameRotate
[03/24 08:44:53.603782][info][11517][FilterFactory.cpp:34]  - HDRMerge
[03/24 08:44:53.603782][info][11517][FilterFactory.cpp:34]  - IMUCorrector
[03/24 08:44:53.603783][info][11517][FilterFactory.cpp:34]  - IMUFrameReversion
[03/24 08:44:53.603784][info][11517][FilterFactory.cpp:34]  - PixelValueOffset
[03/24 08:44:53.603785][info][11517][FilterFactory.cpp:34]  - PixelValueScaler
[03/24 08:44:53.603789][info][11517][FilterFactory.cpp:34]  - PointCloudFilter
[03/24 08:44:53.603790][info][11517][FilterFactory.cpp:34]  - SequenceIdFilter
[03/24 08:44:53.603790][info][11517][FilterFactory.cpp:34]  - ThresholdFilter
[03/24 08:44:53.627571][error][11517][UsbDeviceEnumerator.cpp:147] Start Orbbec USB detection...
[03/24 08:44:53.630699][error][11517][UsbDeviceEnumerator.cpp:152] Found 6 Devices.
[03/24 08:44:53.630883][info][11517][LinuxUsbPal.cpp:310] Create PollingDeviceWatcher!
[03/24 08:44:53.630915][info][11517][DeviceManager.cpp:19] Current found device(s): (2)
[03/24 08:44:53.630920][info][11517][DeviceManager.cpp:27] 	- Name: Gemini 336, PID: 0x0803, SN/ID: CP7X54P0005Z, Connection: USB2.1
[03/24 08:44:53.630922][info][11517][DeviceManager.cpp:27] 	- Name: Gemini 335, PID: 0x0800, SN/ID: CP156410002H, Connection: USB2.1
[03/24 08:44:53.657867][info][11517][DeviceBase.cpp:491] key: AppSlot,value:1
[03/24 08:44:53.657875][info][11517][DeviceBase.cpp:491] key: HwType,value:R1
[03/24 08:44:53.657878][info][11517][DeviceBase.cpp:491] key: HwVer,value:V1.1
[03/24 08:44:53.657881][info][11517][DeviceBase.cpp:491] key: IspFwVer,value:20241125
[03/24 08:44:53.657883][info][11517][DeviceBase.cpp:491] key: IspNeedVer,value:20241125
[03/24 08:44:53.657886][info][11517][DeviceBase.cpp:491] key: PresetVer,value:1.1.9
[03/24 08:44:53.829758][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: FrameUnpacker
[03/24 08:44:53.829923][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: NoiseRemovalFilter
[03/24 08:44:53.829956][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: DisparityTransform
[03/24 08:44:53.829981][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: DepthResize
[03/24 08:44:53.830009][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: HardwareD2DCorrectionFilter
[03/24 08:44:53.830061][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: FrameUnpacker
[03/24 08:44:53.830100][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: FrameUnpacker
[03/24 08:44:53.830134][info][11517][DeviceManager.cpp:140] Device created successfully! Name: Gemini 336, PID: 0x0803, SN/ID: CP7X54P0005Z FW: 1.4.60
[03/24 08:44:53.830212][info][11517][Pipeline.cpp:35] Pipeline created with device: {name: Gemini 336, sn: CP7X54P0005Z}, @0x55C0D8AF4760
[03/24 08:44:53.845894][info][11517][G330Device.cpp:291] Sensor Depth created! Found 26 stream profiles.
[03/24 08:44:53.854615][info][11517][G330Device.cpp:291] Sensor Color created! Found 60 stream profiles.
[03/24 08:44:53.854811][info][11517][Pipeline.cpp:180] Check and set config done!
[03/24 08:44:53.854916][warning][11517][ObException.cpp:8] Filter@FrameProcessor: config schema is empty, doesn't have any config value
[03/24 08:44:53.854981][warning][11517][FrameProcessor.cpp:334] Execute failure! A libobsensor_exception has occurred!
 - where: FrameProcessor.cpp(334):enableHardwareD2CProcess
 - msg: Filter@FrameProcessor: config schema is empty, doesn't have any config value
 - type: N11libobsensor23invalid_value_exceptionE
[03/24 08:44:53.855000][info][11517][Pipeline.cpp:218] Try to start streams!
[03/24 08:44:53.860389][info][11517][Pipeline.cpp:234] Start streams done!
[03/24 08:44:53.860390][info][11517][Pipeline.cpp:202] Pipeline start done!
[03/24 08:44:53.866392][info][11517][DeviceBase.cpp:491] key: AppSlot,value:1
[03/24 08:44:53.866399][info][11517][DeviceBase.cpp:491] key: HwType,value:R1
[03/24 08:44:53.866402][info][11517][DeviceBase.cpp:491] key: HwVer,value:V1.1
[03/24 08:44:53.866434][info][11517][DeviceBase.cpp:491] key: IspFwVer,value:20241125
[03/24 08:44:53.866437][info][11517][DeviceBase.cpp:491] key: IspNeedVer,value:20241125
[03/24 08:44:53.866439][info][11517][DeviceBase.cpp:491] key: PresetVer,value:1.2.2
[03/24 08:44:54.040779][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: FrameUnpacker
[03/24 08:44:54.040854][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: NoiseRemovalFilter
[03/24 08:44:54.040878][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: DisparityTransform
[03/24 08:44:54.040912][warning][11517][ObException.cpp:8] Invalid filter name, cannot find filter creator for filter name: DepthResize
[... 142 lines cut ...]
[03/24 08:44:59.620887][warning][11531][ObException.hpp:48] Update data size(647295) > data buffer size! (614400)
[03/24 08:44:59.620943][warning][11531][ObLibuvcDevicePort.cpp:441] Execute failure! A libobsensor_exception has occurred!
 - where: ObLibuvcDevicePort.cpp(441):onFrameCallback
 - msg: Update data size(647295) > data buffer size! (614400)
 - type: N11libobsensor16memory_exceptionE
```

**System Environment:**

- OS: Linux
- Architecture: x86_64

**Tool/SDK Version:**

- SDK: 2.2.8

**Device Information:**

- Model: Gemini 335 & 336
- Firmware Version: Both have FW ver. 1.4.60

**Additional Information**
- With profiles mentioned above and changed to 15fps, all channels send data at 15fps.
- Using LibUVC backend. Trying the V4L2 backend results in a crash because `No v4l device found on port: x`
- Problem occurs for both values of `FrameMetadataParsingPath` inside config.xml

zhonghong322, 2025-03-28:
1、Using USB 2.0 to connect 2 devices, the USB bandwidth is insufficient.
2、I will reproduce the USB 2.0 V4L2 crash issue you reported later. Is there any problem with V4L2 under USB 3.0?"

Chris-Pane, 2025-03-31:
To 1:
Why exactly is the USB bandwidth insufficient? Using OrbbecViewer I have determined the needed bandwidth for one camera to be 160 Mbps at most. For two cameras, this would be well below the 384 Mbps limit for USB2 bulk transfer when no isochronous transfer is preset. With tools like qv4l2 for Linux, I can in fact open all 4 streams simultaneously at 30fps while both cameras are connected to one USB2 hub.

To 2:
I am no longer able to reproduce this error.