orbbec/OrbbecSDK_v2 #49: Temperature readout stops

Chris-Pane, 2025-03-25:
**Describe the bug**  
When reading out the camera's temperature in a loop, the camera will eventually be unresponsive and not send any data.

**Reproduction Steps**  
Minimal code example: 
```c++
#include <libobsensor/hpp/Context.hpp>
#include <libobsensor/hpp/Device.hpp>
#include <libobsensor/hpp/Error.hpp>

#include <iostream>
#include <thread>
#include <vector>

using namespace std::chrono_literals;

void readTemp(std::shared_ptr<ob::Device> dev) {
    try {
        DEVICE_TEMPERATURE t{};
        uint32_t len = 0;
        dev->getStructuredData(OB_STRUCT_DEVICE_TEMPERATURE, reinterpret_cast<uint8_t*>(&t), &len);

        if (len != sizeof(DEVICE_TEMPERATURE)) {
            return;
        }

        std::cout << std::setw(5) << t.cpuTemp << "  " << t.ldmTemp << "  " << t.rgbTemp << "  "
                  << t.irLeftTemp << "  " << t.irRightTemp << "  ";

    } catch (ob::Error& err) {
        std::cout << "Read error: " << err.what() << std::endl;
        return;
    }
}

int main() {
    ob::Context ctx{};
    std::shared_ptr<ob::DeviceList> devList = ctx.queryDeviceList();
    std::vector<std::shared_ptr<ob::Device>> devices{};

    for (size_t i = 0; i < devList->deviceCount(); ++i) {
        devices.push_back(devList->getDevice(i));
    }

    while (true) {
        for (size_t i = 0; i < devices.size(); ++i) {
            readTemp(devices[i]);
        }
        std::cout << std::endl;
        std::this_thread::sleep_for(100ms);
    }
    return 0;
}
```
This problem occurs with one or more cameras connected via USB 2.1, using UVC backend.

**Log Output**  
```
[03/25 08:59:46.389875][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.392447][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.493609][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.496112][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.597338][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44 
[03/25 08:59:46.599800][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.701000][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.703443][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.804422][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.806795][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.907899][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:46.910291][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.011366][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.013925][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.115118][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.117667][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.218847][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.221272][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.322426][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.324745][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.425905][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.428398][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.529391][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.531622][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:47.632744][debug][4769][PropertyServer.cpp:254] Property 1003 get structure data successfully, size 44
[03/25 08:59:48.438141][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 08:59:58.439159][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 08:59:58.439197][debug][4775][GlobalTimestampFitter.cpp:216] GlobalTimestampFitter update: coefficientA = 999.9881377823758, constantB = 1742883299657604 [**5 logs in 50005ms**]
[03/25 08:59:58.661570][warning][4776][ObException.cpp:8] Resource busy! You can try again later!
[03/25 09:00:06.655677][debug][4870][GlobalTimestampFitter.cpp:216] GlobalTimestampFitter update: coefficientA = 999.988053141125, constantB = 1742883295974673.2 [**3 logs in 58003ms, last: 08:59:38.661100**]
[03/25 09:00:08.440313][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 09:00:08.712306][warning][4776][ObException.cpp:8] Resource busy! You can try again later!
[03/25 09:00:18.441467][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 09:00:18.762851][warning][4776][ObException.cpp:8] Resource busy! You can try again later!
[03/25 09:00:28.442521][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 09:00:28.813437][warning][4776][ObException.cpp:8] Resource busy! You can try again later!
[03/25 09:00:38.443537][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 09:00:38.864034][warning][4776][ObException.cpp:8] Resource busy! You can try again later!
[03/25 09:00:48.444605][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 09:00:48.914591][warning][4776][ObException.cpp:8] Resource busy! You can try again later!
[03/25 09:00:48.914666][debug][4776][GlobalTimestampFitter.cpp:146] The device time RTT has reached the upper limit several times. Sleep for 10000ms and retry
[03/25 09:00:58.445649][debug][4775][PropertyServer.cpp:254] Property 1037 get structure data successfully, size 16
[03/25 09:00:58.445684][debug][4775][GlobalTimestampFitter.cpp:216] GlobalTimestampFitter update: coefficientA = 999.989281977975, constantB = 1742883299650499 [**6 logs in 60006ms**]
```

**System Environment:**

- OS: Linux
- Architecture: x86_64

**Tool/SDK Version:**

- SDK: 2.2.8

**Device Information:**

- Model: Gemini 336
- Firmware Version: 1.4.60

**Additional context**  
- A longer delay inside the loop does not fix the issue
- This issue can occur at any time. It sometimes may take a few minutes
- The device cannot get out of this state. I have noticed multiple `Resource busy` messages when switching streams, these however fix themselves after a small delay (~1 sec)
- Tracing the calls back, I find that `Device::getStructuredData` will eventually call `uvc_set_ctrl` which calls `libusb_control_transfer` with an infinite timeout. I suspect that libusb does not return

Chris-Pane, 2025-03-28:
Adding a timeout of, say 1 second in `uvc_set_ctrl` works around the problem, since you conveniently wrapped `ObLibuvcDevicePort::sendAndReceive` in a loop of multiple retries. However, I don't understand why libusb would sometimes timeout / not respond.

zhonghong322, 2025-03-28:
I will reproduce the issue you reported