orbbec/OrbbecSDK_v2 #53: Don't throw in FrameBufferManagerBase::acquireBuffer

Abrahamh08, 2025-04-12:
https://github.com/orbbec/OrbbecSDK_v2/blob/4b5d54a9b460e028573d3354d1733f61d6c26d63/src/core/frame/FrameBufferManager.cpp#L120

Exceptions cannot be caught easily from this, since frames are aggregated at the low level. It is normal during heavy CPU usage for the buffer to be full for a small while, so frames should just be dropped like how they are at the higher level:

https://github.com/orbbec/OrbbecSDK_v2/blob/4b5d54a9b460e028573d3354d1733f61d6c26d63/src/pipeline/Pipeline.cpp#L259

instead of panicking, which exits my application.

sirudoi, 2025-04-15:
Hi, our latest release package, version 2.3.5, has resolved this issue. Thank you once again for bringing this to our attention. We truly appreciate your contribution, and if you have any further feedback, please feel free to let us know. Otherwise, we plan to close this issue in a few days. Thank you for your understanding and support.

Best regards.

sirudoi, 2025-04-15:
Thank you for your feedback. In the next SDK version, we plan to optimize the internal exception handling—including the issues you mentioned. We appreciate your suggestions and will notify you once the improvements are complete.

Abrahamh08, 2025-04-15:
Great👍

I will say what was causing this issue for me was not heavy CPU usage but the fact that I sleep for a millisecond in my frame processing thread, and when the pipeline is restarted (stop then start a few seconds later), it seems that several frames aggregate at once, which doesn't happen when started for the first time on boot. It's not an issue, though, because I removed the sleep so it handles all those frames very fast after the second start, so we never need to drop.

This doesn't change the need for restructuring, but just a note that my specific issue was not due to heavy CPU usage. But I can see it happening if the system freezes for a bit ever or something like that.

closed 2025-04-15 by Abrahamh08