orbbec/OrbbecSDK_v2 #54: 335 turn on hardware D2D failed

chenglong-do, 2025-04-15:
**Code**  

```c++
  if (device->isPropertySupported(OB_PROP_DISPARITY_TO_DEPTH_BOOL, OB_PERMISSION_WRITE)) {
    device->setBoolProperty(OB_PROP_DISPARITY_TO_DEPTH_BOOL, true);
    std::cout << "turn off software disparity to depth converter (Turn on hardware D2D)" << std::endl;
  } else {
    std::cout << "turn on software disparity to depth converter" << std::endl;
  }
```

**Log Output**  

```
function:ob_device_set_bool_property
args:device:0xaaaaaab73b80, property_id:85, value:1 
message:Component 8 not found!
type:4
```

**System Environment:**

- OS: Ubuntu 22.04
- Architecture: arm64
- GCC：Version 11.4.0

**Tool/SDK Version:**

- SDK: Release v2.3.5

**Device Information:**

- Model: 335
- Firmware Version: 1.4.60

sirudoi, 2025-04-16:
Thank you for your question; it appears that the issue is caused by the SDK’s extension libraries not being loaded correctly.

When you extract the release package(`OrbbecSDK_v2.3.5_202504081238_5c5d8046_linux_aarch64.zip`), you will find an `extensions` folder inside the `OrbbecSDK_v2.3.5_xxx_aarch64/lib` directory. Please ensure that your application's working directory is the same as the directory containing this folder—this should resolve your issue.

Alternatively, you can customize the loading location of the extensions library by using the following interface:
```C++
ob::Context::setExtensionsDirectory("/user/your_directory/extensions");
```

Note: The call to `setExtensionsDirectory` must be made before creating the `Context` object and should be invoked at the class level; otherwise, the configured `extensionsDirectory` will be overridden, as shown in the examples below:
```C++
// Correct usage:
ob::Context::setExtensionsDirectory("/user/your_directory/extensions");
ob::Context context;

// Incorrect usage:
ob::Context context;
context.setExtensionsDirectory("/user/your_directory/extensions");
```

chenglong-do, 2025-04-16:
**Thank you! It's working now, but there is a new issue:**

**With V1 Configuration:** The depth map and color image had consistent resolutions (both 1080p).
```c++
// create config to configure pipeline
    const auto config = std::make_shared<ob::Config>();
    device = pipeline.getDevice();
    // pipeline.enableFrameSync();

    if (device->isPropertySupported(OB_PROP_DISPARITY_TO_DEPTH_BOOL, OB_PERMISSION_WRITE)) {
      device->setBoolProperty(OB_PROP_DISPARITY_TO_DEPTH_BOOL, true);
      std::cout << "turn off software disparity to depth converter (Turn on hardware D2D)" << std::endl;
      USER_LOG_DEBUG("turn off software disparity to depth converter (Turn on hardware D2D)");
    } else {
      std::cout << "turn on software disparity to depth converter" << std::endl;
      USER_LOG_DEBUG("turn on software disparity to depth converter");
    }

    std::shared_ptr<ob::VideoStreamProfile> colorProfile;
    const auto colorProfileList = pipeline.getStreamProfileList(OB_SENSOR_COLOR);
    if (colorProfileList->count() > 0) {
      // get default (index=0) stream profile
      const auto profile = colorProfileList->getVideoStreamProfile(RGB_IMG_WIDTH, RGB_IMG_HEIGHT, OB_FORMAT_RGB, 30);

      colorProfile = profile->as<ob::VideoStreamProfile>();
      std::cout << "color profile: " << colorProfile->width() << "x" << colorProfile->height() << " @ " << colorProfile->fps() << "fps" << std::endl;
      USER_LOG_DEBUG("color profile: %dx%d @ %dfps", colorProfile->width(), colorProfile->height(), colorProfile->fps());
      // enable color stream
      config->enableStream(colorProfile);
    }

    std::shared_ptr<ob::StreamProfileList> depthProfileList;
    OBAlignMode alignMode = ALIGN_DISABLE;
    do {
      if (colorProfile) {
        // Try find supported depth to color align hardware mode profile
        depthProfileList = pipeline.getD2CDepthProfileList(colorProfile, ALIGN_D2C_HW_MODE);
        if (depthProfileList->count() > 0) {
          alignMode = ALIGN_D2C_HW_MODE;
          std::cout << "depth to color align mode: ALIGN_D2C_HW_MODE" << std::endl;
          USER_LOG_DEBUG("depth to color align mode: ALIGN_D2C_HW_MODE");
          break;
        }

        // Try find supported depth to color align software mode profile
        depthProfileList = pipeline.getD2CDepthProfileList(colorProfile, ALIGN_D2C_SW_MODE);
        if (depthProfileList->count() > 0) {
          alignMode = ALIGN_D2C_SW_MODE;
          std::cout << "depth to color align mode: ALIGN_D2C_SW_MODE" << std::endl;
          USER_LOG_DEBUG("depth to color align mode: ALIGN_D2C_SW_MODE");
          break;
        }
      }

      depthProfileList = pipeline.getStreamProfileList(OB_SENSOR_DEPTH);
    } while (false);

    if (depthProfileList->count() > 0) {
      std::shared_ptr<ob::StreamProfile> profile;

      try {
        // Select the profile with the same frame rate as color.
        if (colorProfile) {
          profile = depthProfileList->getVideoStreamProfile(DEPTH_ORIGIN_WIDTH, DEPTH_ORIGIN_HEIGHT, OB_FORMAT_ANY, 30);
        }
      } catch (...) {
        profile = nullptr;
      }

      if (!profile) {
[... 89 lines cut ...]

  // Query the current camera depth mode
  const auto curDepthMode = device->getCurrentDepthWorkMode();
  std::cout << "current depth work mode: " << curDepthMode.name << std::endl;

  // Get the list of camera depth modes
  const auto depthModeList = device->getDepthWorkModeList();
  std::cout << "depthModeList size: " << depthModeList->getCount() << std::endl;
  for (uint32_t i = 0; i < depthModeList->getCount(); i++) {
    std::cout << "depthModeList[" << i << "]: " << (*depthModeList)[i].name << std::endl;
  }

  // switch depth work mode to default (index=0) mode, user can switch to ohter mode like this.
  device->switchDepthWorkMode((*depthModeList)[0].name);
  std::cout << "switch depth work mode to:" << (*depthModeList)[2].name << std::endl;

  // Create a filter to align depth frame to color frame
  auto depth2colorAlign = std::make_shared<ob::Align>(OB_STREAM_COLOR);

  pipeline->start(config);

```

**Question:** How can I configure V2 to ensure both depth map and color image are 1080p?
**Additional Note:** Using auto depth2colorAlign = std::make_shared<ob::Align>(OB_STREAM_COLOR); results in poor performance.

sirudoi, 2025-04-19:
First, I apologize for the delay in getting back to you. Our Gemini 335 currently does not support using `ALIGN_D2C_HW_MODE` at 1080p resolution. Could you please confirm whether, even when using the v1 SDK, you’re in fact using` ALIGN_D2C_SW_MODE` rather than hardware mode?

If you need 1080p with the v2 SDK, you must fall back to software D2C, for example:
```C++
auto depth2colorAlign = std::make_shared<ob::Align>(OB_STREAM_COLOR);
depth2colorAlign->process(frameset);
```
Enabling `ALIGN_D2C_SW_MODE` in v1 and calling `depth2colorAlign->process()` in v2 both perform the alignment in software and incur a similar performance hit. The only difference is that v1 simply flips a config switch, whereas v2 requires you to instantiate `ob::Align` and invoke `process()` manually.

chenglong-do, 2025-04-21:
OK, got it, thank you!

closed 2025-04-21 by chenglong-do