orbbec/OrbbecSDK_v2 #56: Error getting stream profile list for Astra 2...

drhalftone, 2025-05-07:
This used to work with V1.  I am getting the error message, "Filter@FrameProcessor: config schema is empty, doesn't have any config value."

GentBinaku, 2025-05-12:
You need to run the executable in the same directory as the extension folder

drhalftone, 2025-05-13:
My code works with the femto mega I but not the astra 2. Sent from my iPhone

drhalftone, 2025-05-15:
As I wrote the above post, I realized that this issue might be mute as I can't actually use a profile to set the camera resolution.  So my real issue is that I can't find an example of C code for asking a sensor what it's available resolutions are or how to set the resolution I want before I start acquiring video.