orbbec/OrbbecSDK_v2 #58: Gemini 335Le freezes at random (?)

AndreaNappi00, 2025-05-15:
**Describe the bug**  
When connecting the Gemini 335Le camera to the ethernet through PoE switch the camera works as inteded in the orbbec viewer v2 only for 10 seconds. As soon as the I switch from color visualization to pointcloud the image freezes and no more frames are received.

I also tried all possible combinations of resolutions and frame rates.
My switch is capable of 10000Mb/s.
I also tried to power the camera through a 24V DC power source.

**Reproduction Steps**  
Connect the camera to a PoE switch and try to visualize the colored frames or the point cloud. When I connect other devices (in particular the main local network router) the visualization does not even start. 

**Log Output**  
No relevant log output.

**System Environment:**

- OS: Ubuntu 24.04, Fedora 42, Ubuntu 24.10
- Architecture: x86_64

**Tool/SDK Version:**

- SDK: Version 2.3.5
- OrbbecViewer：2.3.5

AndreaNappi00, 2025-05-16:
Update:
After rebooting the camera directly from the Viewer, the camera started operating again as expected. Connecting the network switch with the local network (which connects to more devices) results in a drop in fps.

zhonghong322, 2025-05-20:
[image]    Please modify the xml configuration file and then capture a log.