orbbec/OrbbecSDK_v2 #62: Reading device intrinsics fails...

drhalftone, 2025-05-18:
I don't know if anyone else is using the C library, but I've written a simple Qt widget that tries to connect to an Orbbec device and read its intrinsics, and I'm not having any luck.  I can't exactly find any examples in C.  So maybe someone else has gotten this to work.  Here is my code.

`    ob_context *context = NULL;
    ob_pipeline *pipeline = NULL;
    ob_stream_profile *depthProfile = NULL;
    ob_stream_profile *colorProfile = NULL;
    ob_stream_profile_list *depthProfiles = NULL;
    ob_stream_profile_list *colorProfiles = NULL;

    // CREATE ERROR OBJECT
    ob_error *error = NULL;

    // create context
    context = ob_create_context(&error);
    if (error){
        qDebug() << QString("Error creating context: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    // GET A LIST OF DEVICES
    ob_device_list *deviceList = ob_query_device_list(context, &error);
    if (error){
        qDebug() << QString("Error getting device list: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    uint32_t numDevices = ob_device_list_device_count(deviceList, &error);
    if (error){
        qDebug() << QString("Error getting number of devices: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    ob_device *device = NULL;
    if (numDevices == 0){
        qDebug() << QString("No devices found!");
        return;
    }

    device = ob_device_list_get_device(deviceList, 0, &error);
    if (error){
        qDebug() << QString("Error creating USB device: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    // Create a pipeline to open the depth stream after connecting the device
    pipeline = ob_create_pipeline_with_device(device, &error);
    if (error){
        qDebug() << QString("Error creating pipeline with device: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    // Create config to configure the resolution, frame rate, and format of the depth stream
    ob_config *config = ob_create_config(&error);
    if (error){
        qDebug() << QString("Error creating configuration: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    // CREATE THE DEPTH PROFILE
    depthProfile = ob_create_video_stream_profile(OB_STREAM_DEPTH, OB_FORMAT_Y16, 800, 600, 30.0, &error);
    if (error) {
        qDebug() << QString("Error creating depth profile: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }

    ob_camera_intrinsic intrinsics = ob_video_stream_profile_get_intrinsic(depthProfile, &error);
    if (error) {
        qDebug() << QString("Error getting depth profile intrinsics: %1").arg(error->message);
        ob_delete_error(error);
        error = NULL;
    }
`

zhonghong322, 2025-05-20:
ob_stream_profile_list *stream_profile_list = ob_pipeline_get_stream_profile_list(pipeline, OB_SENSOR_DEPTH, &error);

    depthProfile = ob_stream_profile_list_get_video_stream_profile(stream_profile_list, 800, 600, OB_FORMAT_Y16, 30, &error);

    ob_config_enable_stream(config, OB_STREAM_DEPTH, &error);

    ob_camera_intrinsic intrinsics = ob_video_stream_profile_get_intrinsic(depthProfile, &error);
    if(error) {
        ob_delete_error(error);
        error = NULL;
    }
    printf("fx:fy:cx:cy: %f %f %f %f\n", intrinsics.fx, intrinsics.fy, intrinsics.cx, intrinsics.cy);

zhonghong322, 2025-05-20:
Or getthe stream_profile through the sensor, and then retrieve the intrinsic parameters.

ob_sensor_list *sensor_list = ob_device_get_sensor_list(device, &error);
     if(error) {
         ob_delete_error(error);
         error = NULL;
     }

     ob_sensor *sensor = ob_sensor_list_get_sensor_by_type(sensor_list, OB_SENSOR_DEPTH, &error);

     if(error) {
         ob_delete_error(error);
         error = NULL;
     }

     ob_stream_profile_list *stream_profile_list = ob_sensor_get_stream_profile_list(sensor, &error);
     if(error) {
         ob_delete_error(error);
         error = NULL;
     }

     depthProfile = ob_stream_profile_list_get_video_stream_profile(stream_profile_list, 800, 600, OB_FORMAT_Y16, 30, &error);

     ob_camera_intrinsic intrinsics = ob_video_stream_profile_get_intrinsic(depthProfile, &error);
     if(error) {
         ob_delete_error(error);
         error = NULL;
     }
     printf("fx:fy:cx:cy: %f %f %f %f\n", intrinsics.fx, intrinsics.fy, intrinsics.cx, intrinsics.cy);