orbbec/OrbbecSDK_v2 #72: Hole filter freezes pipeline (astra 2)

JC3, 2025-05-31:
**Describe the bug**  
When the depth hole filter is enabled, `Pipeline::waitForFrames` always times out (even for long timeout parameters, > 10 seconds).

**Reproduction Steps**  

Enable hole filter:

```
device->setBoolProperty(OB_PROP_DEPTH_HOLEFILTER_BOOL, false);
```

**Expected behavior**  

Frames received.

**Log Output**  

No error messages produced.

**System Environment:**

- OS: Ubuntu Jammy, Debian Buster
- Architecture: x86_64

**Tool/SDK Version:**

- SDK: 2.4.3

**Device Information:**

- Model: Astra 2
- Firmware Version: 2.8.20

zhonghong322, 2025-06-01:
Thank you for the feedback. The OB_PROP_DEPTH_HOLEFILTER_BOOL property will be removed in later versions of Astra2 for OrbbecSDK v2.

How to use the Hole Filter refer to: 
https://github.com/orbbec/OrbbecSDK_v2/tree/main/examples/3.advanced.post_processing.

JC3, 2025-06-01:
Thanks for this!

closed 2025-06-01 by JC3