orbbec/OrbbecSDK_v2 #77: "Timestamp anomaly detected" and continuous frame timeouts

JC3, 2025-06-06:
**Describe the bug**  

After running our software with an Astra 2 for approximately 24 hours, calls to `Pipeline::waitForFrames` started returning null after timing out, and messages about "timestamp anomalies" in the Color stream started printing frequently to the logs.

No increased CPU or memory usage was observed while this was happening.

Restarting our software caused everything to return to normal. I will be running it continuously for the next few days to see if it happens again.

**Expected behavior**  

No timeouts, no log messages.

**Log Output**  

Snippet from logs, note that "input_stream" is our software:

```
Jun 06 07:44:23 flipdisc circusd[683]: [06/06 07:44:23.483810][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155874953149, sensor: Color
Jun 06 07:44:23 flipdisc circusd[683]: [06/06 07:44:23.887725][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155875353149, sensor: Color
Jun 06 07:44:24 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:25 flipdisc circusd[683]: [06/06 07:44:25.124984][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155876586482, sensor: Color
Jun 06 07:44:25 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:26 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:26 flipdisc circusd[683]: [06/06 07:44:26.555072][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155878019815, sensor: Color
Jun 06 07:44:27 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:28 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:28 flipdisc circusd[683]: [06/06 07:44:28.754007][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155880219815, sensor: Color
Jun 06 07:44:29 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:29 flipdisc circusd[683]: [06/06 07:44:29.720649][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155881186482, sensor: Color
Jun 06 07:44:31 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:31 flipdisc circusd[683]: [06/06 07:44:31.887407][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155883353149, sensor: Color
Jun 06 07:44:32 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:33 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:33 flipdisc circusd[683]: [06/06 07:44:33.622452][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155885086482, sensor: Color
Jun 06 07:44:33 flipdisc circusd[683]: [06/06 07:44:33.950621][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155885419815, sensor: Color
Jun 06 07:44:34 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:34 flipdisc circusd[683]: [06/06 07:44:34.987766][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155886453149, sensor: Color
Jun 06 07:44:35 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:35 flipdisc circusd[683]: [06/06 07:44:35.520646][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155886986482, sensor: Color
Jun 06 07:44:36 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:36 flipdisc circusd[683]: [06/06 07:44:36.921232][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155888386482, sensor: Color
Jun 06 07:44:37 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:38 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:38 flipdisc circusd[683]: [06/06 07:44:38.889128][error][935][SensorBase.cpp:351] Timestamp anomaly detected, frame: 155890353149, sensor: Color
Jun 06 07:44:39 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
Jun 06 07:44:40 flipdisc circusd[683]: input_stream: WARNING: Timed out waiting for frame.
```

**System Environment:**

- OS: Ubuntu 22.04
- Architecture: x86_64

**Tool/SDK Version:**

- SDK: Version 2.4.3

**Device Information:**

- Model: Astra 2
- Firmware Version: 2.8.20
 
**Additional context**  

Here are some snippets from our software to show the options and things we are using:

```c++
  ob::Pipeline pipeline;

  // ...

  auto device = pipeline.getDevice();
  device->setBoolProperty(OB_PROP_DEPTH_MIRROR_BOOL, true);
  device->setBoolProperty(OB_PROP_COLOR_MIRROR_BOOL, true);
  device->setBoolProperty(OB_PROP_BRT_BOOL, false); // if true will cause low fps
  device->setBoolProperty(OB_PROP_DEPTH_NOISE_REMOVAL_FILTER_BOOL, true);
  device->setIntProperty(OB_PROP_DEPTH_NOISE_REMOVAL_FILTER_MAX_DIFF_INT, 256);
  device->setIntProperty(OB_PROP_DEPTH_NOISE_REMOVAL_FILTER_MAX_SPECKLE_SIZE_INT, 1000);

  // ...

  std::shared_ptr<ob::Config> config(new ob::Config());
  config->disableAllStream();
  config->enableStream(colorProfile);  // it's 800x600x30, RGB
  config->enableStream(depthProfile);  // it's 800x600x30, Y16
  config->setAlignMode(ALIGN_D2C_HW_MODE);
  config->setFrameAggregateOutputMode(OB_FRAME_AGGREGATE_OUTPUT_FULL_FRAME_REQUIRE);
  config->setDepthScaleRequire(true);
  pipeline.enableFrameSync(); // It's enabled by default; but just being explicit.
  pipeline.start(config);

  // ...

  while (true) {

    auto frameSet = pipeline.waitForFrames();
    if (!frameSet) {
      Logger::warning << "Timed out waiting for frame." << std::endl;
      continue;
    }

    auto newColorFrame = frameSet->colorFrame();
    auto newDepthFrame = frameSet->depthFrame();

    // ...

  }
```

zhonghong322, 2025-06-07:
In the v2.4.3 SDK version, a check for abnormal timestamps has been added. If the timestamp is abnormal, the frame will be discarded. Please set the log level to Debug and give me the logs

JC3, 2025-06-07:
@zhonghong322 Ok. Unfortunately I just ran it for 24 hours and the problem did not appear. However, I will enable debug logging and let it run all weekend and hope the problem appears again. This one might be hard to track down. Fingers crossed.

JC3, 2025-06-08:
All right, it happened again and I had Debug logging enabled. Here is about 15 minutes of syslog starting at the first unusual log message before the problem began:

[anomaly-syslog.txt](https://github.com/user-attachments/files/20644418/anomaly-syslog.txt)

Here is the same log that I cleaned up a little (removed non-SDK messages and SDK frame rate reports) if it's useful:

[anomaly-syslog-cleanup.txt](https://github.com/user-attachments/files/20644443/anomaly-syslog-cleanup.txt)

The issue seems to start off like this:

```
[06/07 19:05:47.350221][warning][27778][FormatConverterProcess.cpp:361] Failed to decode mjpeg frame to rgb! Corrupt JPEG data: premature end of data segment
[06/07 19:05:47.402829][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=10.534MB, max limit=2048.000MB
[06/07 19:05:47.436220][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=11.450MB, max limit=2048.000MB
[06/07 19:05:47.469637][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=12.366MB, max limit=2048.000MB
[06/07 19:05:47.502855][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=13.282MB, max limit=2048.000MB
[06/07 19:05:47.536148][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=14.198MB, max limit=2048.000MB
[06/07 19:05:47.569595][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=15.114MB, max limit=2048.000MB
[06/07 19:05:47.602918][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=16.030MB, max limit=2048.000MB
[06/07 19:05:47.636193][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=16.946MB, max limit=2048.000MB
[06/07 19:05:47.669759][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=17.862MB, max limit=2048.000MB
[06/07 19:05:47.702703][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=18.778MB, max limit=2048.000MB
[06/07 19:05:47.736266][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=19.694MB, max limit=2048.000MB
[06/07 19:05:47.769716][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=20.610MB, max limit=2048.000MB
[06/07 19:05:47.803041][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=21.526MB, max limit=2048.000MB
[06/07 19:05:47.836395][debug][27774][FrameBufferManager.cpp:78] New frame buffer allocated=0.916MB, total usage: allocated=22.442MB, max limit=2048.000MB
[06/07 19:05:48.320223][warning][27652][Pipeline.cpp:280] Wait for frame timeout, you can try to increase the wait time! current timeout=1000
[06/07 19:05:50.355500][warning][27778][FormatConverterProcess.cpp:361] Failed to decode mjpeg frame to rgb! Corrupt JPEG data: premature end of data segment [**90 logs in 3005ms**]
[06/07 19:05:52.320725][warning][12435][Pipeline.cpp:280] Wait for frame timeout, you can try to increase the wait time! current timeout=1000 [**3 logs in 4000ms, last: 19:05:51.320790**]
[06/07 19:05:55.116541][warning][27778][ObException.cpp:8] Timestamp anomaly detected, timestamp: 283166510460, cacheTimestamp: 283158710460 ,currentDiff: 7800000, maxValidTimestampDiff: 333330
[06/07 19:05:55.116615][warning][27778][SensorBase.cpp:351] Execute failure! A libobsensor_exception has occurred!
 - where: SensorBase.cpp(351):outputFrame
 - msg: Timestamp anomaly detected, timestamp: 283166510460, cacheTimestamp: 283158710460 ,currentDiff: 7800000, maxValidTimestampDiff: 333330
 - type: N11libobsensor23invalid_value_exceptionE
[06/07 19:05:55.116626][error][27778][SensorBase.cpp:351] Timestamp anomaly detected, frame: 283166510460, sensor: Color
[06/07 19:05:56.387404][warning][27778][FormatConverterProcess.cpp:361] Failed to decode mjpeg frame to rgb! Corrupt JPEG data: premature end of data segment [**180 logs in 6031ms**]
[06/07 19:05:58.321182][warning][12440][Pipeline.cpp:280] Wait for frame timeout, you can try to increase the wait time! current timeout=1000 [**6 logs in 6000ms, last: 19:05:57.321693**]
```

After that happens, for a while, this message seems to appear frequently:

```
[06/07 19:06:08.420238][warning][27778][FormatConverterProcess.cpp:361] Failed to decode mjpeg frame to rgb! Corrupt JPEG data: premature end of data segment [**360 logs in 12032ms**]
```

But eventually that message stops and it's just the timestamp anomalies.

I hope this is useful. I will continue running the software and if it happens again I will let you know if the logs look similar or not.

JC3, 2025-06-09:
@zhonghong322 You asked for a complete log; but I don't see your comment any more. If you still want it: [syslog.2.gz](https://github.com/user-attachments/files/20655832/syslog.2.gz). Software starts at Jun  7 09:16:19.

JC3, 2025-06-12:
@zhonghong322 Is there anything else I can do to help? I have been running our software for about 4 days now but haven't seen the issue appear again yet.

zhonghong322, 2025-06-15:
@JC3 I reviewed the log you provided, and the cause of this issue should be an MJPEG encoding error from the camera, which leads to a decoding failure within the SDK. You can try outputting Color YUYV format for verification.

JC3, 2025-06-15:
@zhonghong322 Yes, it was started by an MJPEG error from the camera. That was obvious from the logs.

What is not obvious is why the SDK couldn't recover after an MJPEG error. Like I said, restarting the software fixes the issue. Logically this means the SDK was stuck in a strange state that it could not recover from, which, as I'm sure you agree, shouldn't happen.

**Why couldn't the SDK resume normal processing after a frame error?** Why did it continue thinking the timestamps were anomalous? Surely a single frame error shouldn't permanently ruin all following timestamps so does that mean there is a bug in the timestamp anomaly detection?

Regardless of the original cause of the frame error, the SDK should not completely fail after it encounters a bad frame.

**Additionally, why was there an MJPEG error in the first place?** Is there a bug in the camera firmware? This should probably be addressed as well.

As for outputting YUYV data: Currently we are already requesting RGB data, what would using YUYV tell us? What should we look for? Given that this failure has only occurred twice in approximately 9 days, the rarity of it makes it difficult to tell if YUYV makes a difference. **Are you suggesting that MJPEG and RGB data are unreliable and that we should prefer YUYV for stability?**

For now, I will have to implement a workaround that restarts our software automatically after a frame timeout. Obviously, this is not ideal, and doesn't seem like the intended design goal of the SDK. Don't you agree?

JC3, 2025-06-15:
Just to be clear, in case it wasn't clear above: When this problem occurs, we are unable to receive *any* frames from the SDK until the software is restarted. It is not an error in just a single frame. The error occurs, and corrupts the stream permanently until software restart.

JC3, 2025-06-16:
All right, this makes sense. I will continue with our workaround, and I will follow up with our support contact.

Thanks for looking into it!

elbal, 2025-11-24:
I am facing a similar issue when using three Femto Bolt, synched throug the Sync Hub Pro and managed by different processes (one for each device). Usually it disappears after resetting power but it resurfaces quite often.
Overall it seems the devices need at least 5 second between their sequential startup or the issue worsen. On the other hand, once they start working for a little bit the issue does not seem to occur later.

I am running the devices through the OrbbecSDK-K4A-Wrapper called through Python.
The specific library I am using is pyKinectAzure (https://github.com/ibaiGorordo/pyKinectAzure), since I want to recover the body skeleton, modified to manage multiple devices concurrently.
I don't think the problem is in Python since the library is mostly doing dll calls.

The devices are updated (1.2.2) and I am runnig the last version of OrbbecSDK_v2 (2.5.5) on Windows 11 Home.

Here is the Log at a debug level:
[OrbbecSDK.log.txt](https://github.com/user-attachments/files/23699930/OrbbecSDK.log.txt)

Hope it helps.