orbbec/OrbbecSDK_v2 #82: VScode supported versions

polyphase-studios, 2025-06-20:
FYI-- VSCode from March 2025 and later are no longer supported (see [link](https://code.visualstudio.com/updates/v1_99#_remote-development))

February 2025 [release](https://code.visualstudio.com/updates/v1_98) confirmed to work on Mac OS host

simbachou, 2025-06-21:
Hi! I noticed you mentioned that VSCode from March 2025 and later isn’t supported.
However, I’m currently using the latest VSCode on macOS and everything works fine.
Could you share any logs or error messages that explain the incompatibility? I’d like to help verify or troubleshoot if needed.

Thanks!

polyphase-studios, 2025-06-23:
When using May 2025 release, I am receiving the message "Could not establish connection to "192.168.0.10": The remote host does not meet the prerequisites for running VS Code Server." when connecting using ssh.

Also, whenever I restart the Femto Mega, I have to clear my local known hosts since the ssh signature changes. Otherwise connecting using ssh from the terminal works as expected.

simbachou, 2025-06-24:
Thanks for the feedback.

Could you please share more details about your setup?
1. What device are you using? If it's a Femto Mega, what is the device IP?
2. Which application or tool are you using (e.g., sample using OrbbecSDK), and what operation are you trying to perform

This information will help us better understand the issue. Thanks!

polyphase-studios, 2025-06-24:
Connecting to Femto Mega through Ethernet, device ip 192.168.0.10 in Developer Mode.
We are using firmware 1.3.1 and v2-mega-ipc branch to develop our application.
Currently we are having issues using setDepthScaleRequire when using HW D2C align. Will create a separate bug for more details.