orbbec/OrbbecSDK_v2 #84: color intrinsics and distortion parameters

bhoram, 2025-07-17:
This is not an issue but a question.

It seems by default RGB images are rectified (undistorted). 
- Is this true? 
- If true, then what are distortion parameters for?
- If not true, is there a built-in undistort function I can call? 

(I am playing with Gemini 2)

zhonghong322, 2025-08-05:
The raw color image from Gemini 2 has distortion, but the distortion is minimal. If you need to undistort the image, the distortion parameters can be obtained through the stream profile. The SDK does not provide undistortion function.

The interface to obtain the distortion parameters is as follows:

https://orbbec.github.io/docs/OrbbecSDKv2_API_User_Guide/source/3_Application_Guide/ApplicationGuide.html#obtain-intrinsic-and-extrinsic-parameters

schwer2358, 2025-09-03:
How is this for the other sensors?