orbbec/OrbbecSDK_v2 #85: Can't get stream profile list for OB_SENSOR_IR and OB_SENSOR_DEPTH

LiubovPil, 2025-07-23:
**Describe the bug**  
Hi there!

I have installed OrbbecSDK 2.4.8 to get ir and depth streams from Orbbec Femto Bolt (this camera has 5 type of stream - OB_SENSOR_IR, OB_SENSOR_COLOR, OB_SENSOR_DEPTH, OB_SENSOR_ACCEL, OB_SENSOR_GYRO)
When I tried to get stream profile list for OB_SENSOR_IR and OB_SENSOR_DEPTH types I got errors - "DeviceComponentPtr is nullptr", also I met another errors (like "Component 28 not found!") if I try to get stream profile list after get "DeviceComponentPtr is nullptr" error first. Builded examples don't launch (I guess for the same reason)

For OB_SENSOR_COLOR, OB_SENSOR_ACCEL and OB_SENSOR_GYRO there is no such error as well there is no error in OrbbecViewer. In OrbbecSDK 1.10.22 I can get stream profile list for ir and depth types with no error, too

**Reproduction Steps**  
Code:

```
#define _CRT_SECURE_NO_WARNINGS

#include <iostream>
#include <memory>
#include <exception>

#include <libobsensor/ObSensor.hpp>

int main()
{
    ob::Pipeline pipe;
    std::shared_ptr<ob::Device> device = pipe.getDevice();

    std::shared_ptr<ob::DeviceInfo> deviceInfo = device->getDeviceInfo();

    std::cout << "DeviceInfo - name - " << deviceInfo->getName() << " - type - " <<
        deviceInfo->getConnectionType() << " - firmwareVersion - " << deviceInfo->getFirmwareVersion() << std::endl;

    std::shared_ptr<ob::Config> config = std::make_shared<ob::Config>();
    
    try {
        std::shared_ptr<ob::StreamProfileList> depthProfiles = pipe.getStreamProfileList(OBSensorType::OB_SENSOR_DEPTH);
        std::cout << "Depth profiles count: " << depthProfiles->count() << std::endl;
    }
    catch (std::exception& e) {
        std::cout << "Error - " << e.what() << std::endl;
    }

    try {
        std::shared_ptr<ob::StreamProfileList> irProfiles = pipe.getStreamProfileList(OBSensorType::OB_SENSOR_IR);
        std::cout << "IR profiles count: " << irProfiles->count() << std::endl;
    }
    catch (std::exception& e) {
        std::cout << "Error - " << e.what() << std::endl;
    }
}
```

**Expected behavior**  
I want to get OB_SENSOR_IR and OB_SENSOR_DEPTH streams via code not OrbbecViewer  

**Log Output**  
```
DeviceInfo - name - Orbbec Femto Bolt - type - USB3.1 - firmwareVersion - 1.1.2
Error - DeviceComponentPtr is nullptr
Error - Component 28 not found!
```

**System Environment:**

- OS: Windows 11
- Architecture: x64
- Visual Studio/GCC：Visual Studio Community 2022 17.13.4

**Tool/SDK Version:**

- SDK: OrbbecSDK 2.4.8
- OrbbecViewer：supplied with OrbbecSDK 2.4.8

**Device Information:**

- Model: Orbbec Femto Bolt
- Firmware Version: 1.1.2

**Additional context**  
Camera was manually upgraded to firmware version 1.1.2

kopeto, 2025-07-24:
Your app needs the "Extensions" folder to be in executable's path. I think copying this [extensions](https://github.com/orbbec/OrbbecSDK_v2/tree/main/extensions) folder in the same path you are executing your app should work.

kopeto, 2025-07-24:
I'm on linux and I don't use the GStreamer plugin. I can't help you on that, sorry.