orbbec/OrbbecSDK_v2 #89: Gemini 2 dual ir image

lubkwo, 2025-07-30:
Describe the Bug
I initially attempted to retrieve images from all available cameras using your Python SDK. The IR camera on the right side (when the device is facing me) returns a proper IR image. However the left IR camera only provides a processed image not a raw IR image like the right one. I suspected this might be a limitation of the Python SDK, but I encountered the same issue using the C++ SDK.
Interestingly, in OrbbecViewer I can view both the left and right IR camera feeds without issue—but not simultaneously. 

Reproduction Steps
Steps to reproduce the issue:

- Attempt to access all camera streams using the SDK.

- Observe the output of the left and right IR cameras.

Expected Behavior
I expect the left IR camera to return a raw IR image in the same format as the right IR camera not a processed version.

System Environment:

    OS: Ubuntu 24.04.2 LTS

    Architecture: x86_64

Tool/SDK Version:

    SDK: 2.4.8

Device Information:

    Model: Orbbec Gemini 2

    Firmware Version: 1.4.92

zhonghong322, 2025-08-05:
[image]   In this depth mode, both the left and right IR images can be retrieved, but the left IR image is not the raw image. Simultaneous retrieval of raw images from both left and right IR cameras is not supported.

lubkwo, 2025-08-06:
Okay, but will I ever be able to do something like that in your SDK? What I mean is, get an image like this: left IR, right IR, and RGB? Do you have any plans to add something like that? Or do you have any idea how I could do it? I would be very grateful for your reply.