orbbec/OrbbecSDK_v2 #92: Depth and color frames not synchronized on Astra 2

oliviamiller, 2025-08-13:
I am using an Orbbec Astra 2 with OrbbecSDK v2.4.3 on macOS, firmware version 2.8.20.

I expected that calling enableFrameSync() would synchronize the depth and color frames in a frameset, but i'm still seeing a few milliseconds of difference between the timestamps. 

Code:
```
std::shared_ptr<ob::Config> createHwD2CAlignConfig(std::shared_ptr<ob::Pipeline> pipe) {
    auto colorStreamProfiles = pipe->getStreamProfileList(OB_SENSOR_COLOR);
    auto depthStreamProfiles = pipe->getStreamProfileList(OB_SENSOR_DEPTH);

    // Iterate through all color and depth stream profiles to find a match for
    // hardware depth-to-color alignment
    auto colorSpCount = colorStreamProfiles->getCount();
    auto depthSpCount = depthStreamProfiles->getCount();
    for (uint32_t i = 0; i < colorSpCount; i++) {
        auto colorProfile = colorStreamProfiles->getProfile(i);
        auto colorVsp = colorProfile->as<ob::VideoStreamProfile>();
        for (uint32_t j = 0; j < depthSpCount; j++) {
            auto depthProfile = depthStreamProfiles->getProfile(j);
            auto depthVsp = depthProfile->as<ob::VideoStreamProfile>();

            // make sure the color and depth stream have the same fps, due to some
            // models may not support different fps
            if (colorVsp->getFps() != depthVsp->getFps()) {
                continue;
            }

            // Check if the given stream profiles support hardware depth-to-color
            // alignment
            if (checkIfSupportHWD2CAlign(pipe, colorProfile, depthProfile)) {
                // If support, create a config for hardware depth-to-color alignment
                auto hwD2CAlignConfig = std::make_shared<ob::Config>();
                hwD2CAlignConfig->enableStream(colorProfile);       // enable color stream
                hwD2CAlignConfig->enableStream(depthProfile);       // enable depth stream
                hwD2CAlignConfig->setAlignMode(ALIGN_D2C_HW_MODE);  // enable hardware depth-to-color alignment
                hwD2CAlignConfig->setFrameAggregateOutputMode(OB_FRAME_AGGREGATE_OUTPUT_ALL_TYPE_FRAME_REQUIRE);  // output
                                                                                                                  // frameset
                                                                                                                  // with all
                                                                                                                  // types of
                                                                                                                  // frames
                return hwD2CAlignConfig;
            }
        }
    }
    return nullptr;
}

int main(void) try {
    ob::Context ctx;
    std::shared_ptr<ob::DeviceList> devList = ctx.queryDeviceList();
    std::shared_ptr<ob::Device> dev = devList->getDevice(0);

    std::shared_ptr<ob::Pipeline> pipe = std::make_shared<ob::Pipeline>(dev);
    pipe->enableFrameSync();
    auto config = createHwD2CAlignConfig(pipe);
    std::shared_ptr<ob::PointCloudFilter> pointCloudFilter = std::make_shared<ob::PointCloudFilter>();
    std::shared_ptr<ob::Align> align = std::make_shared<ob::Align>(OB_STREAM_COLOR);
    pointCloudFilter->setCreatePointFormat(OB_FORMAT_RGB_POINT);

    std::mutex                    framesetMutex;
    std::shared_ptr<ob::FrameSet> frameset = nullptr;

    // Start the pipeline with callback.
    pipe->start(config, [&](std::shared_ptr<ob::FrameSet> frameSet) {
        std::lock_guard<std::mutex> lock(framesetMutex);
        frameset = frameSet;
        if (frameSet->getCount() != 2) {
            std::cerr << "got non 2 frame count: " << frameSet->getCount() << "\n";
        }
        std::shared_ptr<ob::Frame> color = frameSet->getFrame(OB_FRAME_COLOR);
        if (color == nullptr) {
            std::cerr << "no color frame\n" << frameSet->getCount() << "\n";
        }

        std::shared_ptr<ob::Frame> depth = frameSet->getFrame(OB_FRAME_DEPTH);
        if (depth == nullptr) {
            std::cerr << "no depth frame\n";
        }

        std::cout << "color: " << color->getSystemTimeStampUs() << " depth: " <<  depth->getSystemTimeStampUs() << "\n";

    });

```

Output:
```
color: 1755114554305317 depth: 1755114554313262
color: 1755114554338513 depth: 1755114554346645
color: 1755114554371830 depth: 1755114554379914
color: 1755114554405166 depth: 1755114554413286
color: 1755114554438526 depth: 1755114554446662
color: 1755114554471821 depth: 1755114554480019
color: 1755114554505284 depth: 1755114554513883
color: 1755114554538513 depth: 1755114554546647
color: 1755114554571864 depth: 1755114554580043
color: 1755114554605190 depth: 1755114554613379
color: 1755114554638534 depth: 1755114554646682
color: 1755114554671825 depth: 1755114554680043
color: 1755114554705141 depth: 1755114554713388
color: 1755114554738525 depth: 1755114554746725
```

Is there a way to obtain synchronized frame sets on the astra 2?

simbachou, 2025-08-27:
Thanks for your feedback! Just to clarify:

getSystemTimeStampUs() returns the system timestamp (the moment when the frame is received by the OrbbecSDK).
getTimeStampUs() returns the frame’s hardware timestamp (generated by the device itself).

So if you need the actual capture time of the frame, please use getTimeStampUs()

oliviamiller, 2025-08-27:
@simbachou Thanks for the clarification, even when using getTimeStampUs() i'm seeing a few ms of difference between the two frames. 
```
color: 1065702556 depth: 1065708916
color: 1065735889 depth: 1065742258
color: 1065769221 depth: 1065775598
color: 1065802556 depth: 1065808937
color: 1065835889 depth: 1065842278
color: 1065869222 depth: 1065875618

```